FreeCAD/FreeCAD #18280: Expressions in Constraints Sometimes Replaced with Static Values

Hatsuwr, 2024-12-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've had an issue where expressions used in constraint values are sometimes replaced by the computed static value of the expression at the time. I haven't been able to reproduce this issue reliably though.

Two potentially related incidents occurred after the expressions were initially set, although not necessarily before the replacement happened:

-An out-of-memory error caused by accidentally adding an extra digit to the number of occurrences in a polar pattern.
-A few unresolved problems related to overlapping geometry in polar pattern occurrences.

It’s unclear if these incidents are connected to the expression issue, but the pattern in each case was applied to the pad of the sketch that had the constraint expressions replaced.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-05:
@Hatsuwr please add a demo file and steps to reproduce. Do you mean constraints in Assembly or Sketcher WB?

luzpaz, 2025-01-31:
@Hatsuwr 👆

closed 2025-02-22 by maxwxyz

coolbluewater, 2025-09-14:
I hit this bug on my second FreeCAD project! How is this acceptable to close?
Is there something that specifically triggers it and are there any workarounds?
Loss of data is about as severe as a bug can get...