FreeCAD/FreeCAD #18283: Techdraw Rich Text Annotations and Leader Lines Do not maintain position after save.

jlovett11211, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have added several rich text notations and leader lines in my Techdraw page. Each of those notations and lines are associated with an Anno Parent or Leader Parent. (Before saving and closing the objects move in sync with the associated view.) After saving, closing, & reopening, the notations and lines are moved off of the page to the bottom left. The Parent associations are still there, but the objects no longer move in sync with the associated view. If I move the object to the correct location and reset the parent, the object moves with the parent as it should. However upon saving, closing, and reopening, the same thing happens again. This is repeatable in 1.0.0 and in 1.1.0 dev Rev 39380. This also repeats in Safe Mode.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.67
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * freecad.gears 1.2.0
  * Gridfinity 0.5.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

TechDraw

### Anything else?

[p74.248 Horizontal Stirling Engine Plans v03.zip](https://github.com/user-attachments/files/18001965/p74.248.Horizontal.Stirling.Engine.Plans.v03.zip)

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-04:
@WandererFan FYI

WandererFan, 2024-12-05:
I can't reproduce this behaviour.  I see problems in the tree in sample file - there are 2 rta's and 1 leader that are not shown as children of the page even though they are listed in the page's view list. 

I'm running a somewhat newer version.  

OS: Linux Mint 21.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39454 (Git)
Build type: debug
Branch: invLeaderAndRTAPositionAfterSave
Hash: a97677ed2f0ee4875321f3043782cf77e3756cc6
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21

This one saves and restores without problem here.  Does it work for you?
[RTADragTest_save.FCStd.not.zip](https://github.com/user-attachments/files/18015921/RTADragTest_save.FCStd.not.zip)

[image]

WandererFan, 2024-12-05:
@jlovett11211 do you happen to have an FCBak file from just before things went south?

jlovett11211, 2024-12-05:
Thanks for your response @WandererFan  I was not able to replicate the issue with your sample file. 

I think I found the issue / a workaround on my file though. It was still happening after I cleaned up the tree, but thanks for pointing that out anyway. I started looking at the specified parent for the rta's. The ones that were moving were associated with views, not the Part. Once I changed the association to the part, it stopped moving. 

Is there a more appropriate way to associate a rta with a part directly instead of a view? 

(I just saw your response about an earlier save. Unfortunately, this started happening right from the beginning, so I don't have an earlier backup. )

jlovett11211, 2024-12-05:
I've come across another quirk. After adding additional rta's and leaderlines and making sure that the anno parents are all parts, I saved and closed. When I reopened, All of the new rta's and leaderlines stayed in place, but moved outside of the parts on the tree. 

[image]

If I try to drag the rta's into the part, I get the following error:
[image]

WandererFan, 2024-12-05:
@jlovett11211 I think I've got the misplaced rta's figured out.  The messed up leaders are linked to rta's so they should be sorted also. Still no luck in messing up the tree.

Does this look right to you?  All the items that were in the bottom left have moved, but there is a lot going on on the page.

[image]

jlovett11211, 2024-12-05:
That looks pretty close. I kept working on it last night. The moving rta's have stopped since I started parenting them to the main part instead of the view. The rta's still populate at the bottom of the tree, so I've been moving them into a folder.  

I appreciate you looking into this. It's the first time I've reported a bug or really used github so let me know if I'm not doing something right. I've been using FreeCAD on and off since last fall, and I just love it, so I hope that this conversation can be helpful in some way. 

Here's the current version:
[p74.248 Horizontal Stirling Engine Plans v04.FCStd.not.zip](https://github.com/user-attachments/files/18026342/p74.248.Horizontal.Stirling.Engine.Plans.v04.FCStd.not.zip)

I've been using the following build:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.67
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * freecad.gears 1.2.0
  * Gridfinity 0.5.0
  * Manipulator 1.5.7

WandererFan, 2024-12-05:
You're doing a great job as a bug reporter.  Keep it up please!

ProjectionGroups don't know that they can have children that are not views (ProjectionGroupItems).  So that explains the tree issue, I think.

Parenting the rta to the ProjectionGroupItem is supposed to work.  The work around to parent to the whole group should not be necessary.

jlovett11211, 2024-12-05:
Thanks! I'll keep on with my workaround and check in if I learn any more.

commit 93aa3b31 (2025-01-06): [TD]Fix RichText parent in scene (fix #18283) (#18768)

* [TD]allow DrawViewAnno to be attached to another view

* [TD]allow image and spreadsheet attach to view

* [TD]ensure correct parent in scene

* [TD]refactor command helpers to separate file

- gathering the helpers in one place
- helper redundancy to be address in another change

* [TD]create symbol, spreadsheet, image with parent

* [TD]fix claimChildren for some views

- also includes many lint fixes
closed 2025-01-06 by yorikvanhavre