FreeCAD/FreeCAD #18285: STL: Position is not respected when exporting to STL format

WallieEverest, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Problem: Exporting an STL is multiplying an object's "Placement-Position" offset by 2X.

1.) Create a new file with a default 10mm cube part.
2.) Edit the Placement-Position to be non-zero, ex. {1mm, 5mm, 0mm}.
3.) Export as an STL, then import the STL.
4.) The imported cube will be located at {2mm, 10mm, 0mm}, a 2X offset of the original location.

Nothing in Edit-Preferences appears to be related to this problem.

The work around is to downgrade-upgrade the object using the Draft toolbox, clearing the Placement-Position offset and forcing an absolute location.

This issue occurs in versions 0.21.2 and the newer 1.0.0.
[stl-example.zip](https://github.com/user-attachments/files/18002114/stl-example.zip)

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.12
  * Help 1.0.3
  * kicadStepUpMod 11.3.9
  * Render 2024.1.23
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-04:
> Version: 0.21.2.33771 (Git)

Should be fixed in 1.0  
See #18051

closed 2024-12-04 by luzpaz

WallieEverest, 2024-12-05:
Confirmed the issue is isolated to version 0.21.2.