FreeCAD/FreeCAD #18287: [Compile in windows]Build_Mesh_Part is always in OFF, whether I open the option BUILD_MESH_PAET

liquor2349, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Because of that, when I used the FreeCAD to mesh a part, it will error No module named MeshPartGui.
The cmake-gui is as follows:
[image]

### Full version info

```shell
My environment is Windows11, cmake 3.24.2 and visual studio 2022.
I use the the FreeCAD-releases-FreeCAD-1-0 and the libpack is LibPack-1.0.0-v3.0.0RC4-Release
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

liquor2349, 2024-12-04:
I got a solution that check BUILD_FEM, then the mesh part can be changed to ON.
But is it possible to build Mesh Part without FEM workbench? I just want to get the mesh from parts and make the software as lightweight as possible.
Looking forward to your reply. 
Best wishes!

maxwxyz, 2024-12-05:
@marioalexis84 @FEA-eng maybe?

marioalexis84, 2024-12-05:
Can you check if setting `BUILD_FEM_NETGEN=ON` and `BUILD_FEM=OFF` fixes your problem?
MeshPart and Fem are independent modules. You should be able to build MeshPart without Fem.

liquor2349, 2024-12-05:
> Can you check if setting `BUILD_FEM_NETGEN=ON` and `BUILD_FEM=OFF` fixes your problem? MeshPart and Fem are independent modules. You should be able to build MeshPart without Fem.

Thank you for your reply, here is the screenshot of my CMake. I just checked the Build_FEM, without any other change. I also think that MeshPart and Fem are independent modules, but the fact seems to be confusing.

[image]
[image]

PR #18322 (2024-12-09): CMake: Decouple internal SMESH build from Fem build - fixes #18287
Fixes #18287
Add a new variable `FREECAD_USE_SMESH`.
`BUILD_SMESH` is set to `ON` only if the internal SMESH package is needed.
commit a6b36f6e (2024-12-09): CMake: Decouple internal SMESH build from Fem build - fixes #18287
closed 2024-12-09 by chennes