FreeCAD/FreeCAD #18290: Freecad crashes when trying to fillet/chamfer this surface

naddeoa, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad 1.0 crashes whenever I try to chamfer the surface in the screenshot. Its 100% reproducible.

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1202
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.33
  * A2plus 0.4.68
```

### Subproject(s) affected?

None

### Anything else?

Link to the project file: https://github.com/naddeoa/forum-files/blob/master/humidity%20sensor%20holder.FCStd (not sure how to attach it)

Here is the segfault 

```
⟩ snap run freecad
FreeCAD 1.0.0, Libs: 1.0.0R38641 +468 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Adding snap-specific PYTHONPATH to sys.path: /home/anthony/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1202/lib/python3.10/site-packages:/snap/freecad/1202/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x78cc04c00520]
#1  0x78cbebf54564 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, opencascade::handle<Geom_Surface> const&, TopLoc_Location const&, double&, double&, bool*) from /snap/freecad/1202/usr/lib/libTKBRep.so.7.7+0x174
#2  0x78cbebf54642 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, TopoDS_Face const&, double&, double&, bool*) from /snap/freecad/1202/usr/lib/libTKBRep.so.7.7+0xb2
#3  0x78cbebf62a6b in BRepAdaptor_Curve2d::Initialize(TopoDS_Edge const&, TopoDS_Face const&) from /snap/freecad/1202/usr/lib/libTKBRep.so.7.7+0xdb
#4  0x78cbd8bf9a36 in ChFi3d_Builder::StartSol(opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, opencascade::handle<BRepAdaptor_Curve2d>&, double&, opencascade::handle<ChFiDS_SurfData> const&, bool, int, opencascade::handle<BRepAdaptor_Surface>&, opencascade::handle<BRepAdaptor_Curve2d>&, bool&, bool&, bool&, bool&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, bool, TopoDS_Vertex const&) const from /snap/freecad/1202/usr/lib/libTKFillet.so.7.7+0xd06
#5  0x78cbd8bfecab in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_ElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) from /snap/freecad/1202/usr/lib/libTKFillet.so.7.7+0x63b
#6  0x78cbd8c01558 in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) from /snap/freecad/1202/usr/lib/libTKFillet.so.7.7+0x108
#7  0x78cbd8c069a9 in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) from /snap/freecad/1202/usr/lib/libTKFillet.so.7.7+0x69
#8  0x78cbd8bcbb5e in ChFi3d_Builder::Compute() from /snap/freecad/1202/usr/lib/libTKFillet.so.7.7+0x42e
#9  0x78cbd8d5d111 in BRepFilletAPI_MakeChamfer::Build(Message_ProgressRange const&) from /snap/freecad/1202/usr/lib/libTKFillet.so.7.7+0x31
#10  0x78cbe8d03d8d in BRepBuilderAPI_MakeShape::Shape() from /snap/freecad/1202/usr/lib/libTKTopAlgo.so.7.7+0x8d
#11  0x78cbdafbf376 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /snap/freecad/1202/usr/lib/Part.so+0x86
#12  0x78cbdafbf514 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) from /snap/freecad/1202/usr/lib/Part.so+0x84
#13  0x78cbdafc15ff in Part::TopoShape::makeElementChamfer(Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, Part::ChamferType, double, double, char const*, Part::Flip) from /snap/freecad/1202/usr/lib/Part.so+0x39f
#14  0x78cbcc2b6fce in PartDesign::Chamfer::execute() from /snap/freecad/1202/usr/lib/_PartDesign.so+0x2ee
#15  0x78cc06fde5da in App::DocumentObject::recompute() from /snap/freecad/1202/usr/lib/libFreeCADApp.so+0xba
#16  0x78cbdad5b9cd in Part::Feature::recompute() from /snap/freecad/1202/usr/lib/Part.so+0xd
#17  0x78cc06f788ed in App::Document::_recomputeFeature(App::DocumentObject*) from /snap/freecad/1202/usr/lib/libFreeCADApp.so+0x38d
#18  0x78cc06f8021b in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /snap/freecad/1202/usr/lib/libFreeCADApp.so+0x65b
#19  0x78cc0704f658 in App::DocumentPy::recompute(_object*) from /snap/freecad/1202/usr/lib/libFreeCADApp.so+0x2e8
#20  0x78cc0704f79c in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /snap/freecad/1202/usr/lib/libFreeCADApp.so+0x2c
#21  /snap/freecad/1202/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128008) [0x78cc06597008]
#22  /snap/freecad/1202/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x78cc0654eadc]
[... 27 lines cut ...]
#50  /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c91cb) [0x78cc05f031cb]
#51  /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc525) [0x78cc05f06525]
#52  0x78cc05ea6763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#53  0x78cc07a879b0 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1202/usr/lib/libFreeCADGui.so+0xd0
#54  0x78cc0530ce4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#55  0x78cc057351e7 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x687
#56  0x78cc05708dbc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#57  /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5(+0x76f5e) [0x78cbfedd4f5e]
#58  /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x78cc02dc1d3b]
#59  /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x78cc02e17258]
#60  /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x78cc02dbf3e3]
#61  0x78cc05366338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#62  0x78cc0530b76b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#63  0x78cc05313d14 in QCoreApplication::exec() from /snap/freecad/1202/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#64  0x78cc079c2c88 in Gui::Application::runApplication() from /snap/freecad/1202/usr/lib/libFreeCADGui.so+0x6b8
#65  /snap/freecad/1202/usr/bin/FreeCAD(+0x9791) [0x63dc61181791]
#66  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x78cc04be7d90]
#67  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x78cc04be7e40]
#68  /snap/freecad/1202/usr/bin/FreeCAD(+0x9b55) [0x63dc61181b55]

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-04:
Confirm

<details>
  <summary>1.1.0dev.39416 (Git) Conda AppImage, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

PS
The sketch contain the error.
Eliminating the error does not solve the problem, but simply points out what you should not to do.
[image]

naddeoa, 2024-12-09:
@tarman3 thank you.

Can I ask what's wrong with doing that? Is it going to end up protruding through the chamfer?

tarman3, 2024-12-09:
To keep problems to a minimum
- the sketch must be completely limited
- the wires must be closed
- wires should not intersect