FreeCAD/FreeCAD #18293: BIM: Bim_Project command not working

maxwxyz, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've tried to setup an IFC project in the BIM WB but as soon as I click on the command there are a few errors and nothing happens:
[image]

@yorikvanhavre FYI

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-12-04:
I am not able to reproduce the issue. `BIM_Project` creates an empty IFC project inside current FCstd file without any errors.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

maxwxyz, 2024-12-04:
Also confirmed on a different PC with Windows 11 and Windows 10:

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.1
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.31
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.6
  * Silk 0.1.5

```

maxwxyz, 2024-12-04:
@semhustej can you try with the latest weekly too?

semhustej, 2024-12-04:
> @semhustej can you try with the latest weekly too?

I can reproduce. Probably had the problem with the previous version too, but the warning is not displayed in the 'user notification' for me and I did not have report view open previously.

When running in safe mode, I get the same warning as you do.

When running with my standard settings, I get different warning though:
```
14:42:52  Note: API not available due to missing dependencies: geometry.add_door_representation - No module named 'mathutils'
14:42:52  Note: API not available due to missing dependencies: geometry.add_railing_representation - No module named 'mathutils'
14:42:52  Note: API not available due to missing dependencies: geometry.add_representation - No module named 'bpy'
14:42:52  Note: API not available due to missing dependencies: geometry.add_window_representation - No module named 'mathutils'
14:42:52  Note: API not available due to missing dependencies: grid.create_axis_curve - No module named 'mathutils'
14:42:52  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 76, in onChanged
    self.edit_pset(obj, prop)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 215, in edit_pset
    ifc_psets.edit_pset(obj, prop)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_psets.py", line 218, in edit_pset
    ifc_tools.api_run(
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 235, in api_run
    result = ifcopenshell.api.run(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\api\__init__.py", line 92, in run
    return usecase_function(ifc_file, should_run_listeners=should_run_listeners, **settings)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\api\__init__.py", line 252, in wrapper
    result = usecase(*args, **settings)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\api\pset\edit_pset.py", line 166, in edit_pset
    return usecase.execute()
           ^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\api\pset\edit_pset.py", line 174, in execute
    new_props = self.add_new_properties()
      <class 'ValueError'>: invalid null reference in method 'new_IfcBaseClass', argument 2 of type 'std::string const &'
14:42:52  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 76, in onChanged
    self.edit_pset(obj, prop)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_objects.py", line 215, in edit_pset
    ifc_psets.edit_pset(obj, prop)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_psets.py", line 218, in edit_pset
    ifc_tools.api_run(
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 235, in api_run
    result = ifcopenshell.api.run(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\api\__init__.py", line 92, in run
    return usecase_function(ifc_file, should_run_listeners=should_run_listeners, **settings)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\api\__init__.py", line 252, in wrapper
    result = usecase(*args, **settings)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\api\pset\edit_pset.py", line 166, in edit_pset
    return usecase.execute()
           ^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\api\pset\edit_pset.py", line 174, in execute
    new_props = self.add_new_properties()
      <class 'ValueError'>: invalid null reference in method 'new_IfcBaseClass', argument 2 of type 'std::string const &'
14:42:52  pyException: Traceback (most recent call last):
[... cut ...]
    new_props = self.add_new_properties()
      <class 'ValueError'>: invalid null reference in method 'new_IfcBaseClass', argument 2 of type 'std::string const &'

```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

semhustej, 2024-12-04:
I have similar issue when switching to Locked IFC mode.

luzpaz, 2024-12-04:
@tritao full About info please

yorikvanhavre, 2024-12-05:
This is all probably because of the recent "Physical Properties" added by the Materials system. https://github.com/FreeCAD/FreeCAD/pull/18264 contains a fix for that, that I'll merge ASAP (as soon as I fix the conflicts)

yorikvanhavre, 2024-12-05:
> @yorikvanhavre If it helps, the error I was seeing was around 

Oh that's indeed another error... it sould be `ptype` instead of `ifcprop`. I'll fix that in another PR

maxwxyz, 2024-12-20:
@yorikvanhavre @marcuspollio I still have this issue in the latest weekly:

1. New document
2. Click BIM_Project : Error and nothing happens:

[image]

Can you confirm?

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.2
  * Silk 0.1.5

```

semhustej, 2024-12-20:
I can confirm the same error as @maxwxyz is posting.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

yorikvanhavre, 2024-12-23:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/18011

PR #18312 (2024-12-09): BIM: Fixed copy-paste bug in nativeifc psets
Fixes #18293
commit 29ea9717 (2024-12-09): BIM: Fixed copy-paste bug in nativeifc psets - Fixes #18293
closed 2025-01-07 by maxwxyz