FreeCAD/FreeCAD #18295: BIM: opening BIMExample on latest weekly crashes FreeCAD

maxwxyz, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. Open BIMExample from Start page
2. FreeCAD crashes without notice

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.1
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.31
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.6
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-04:
@yorikvanhavre FYI
@semhustej can you try on the latest weekly too?

yorikvanhavre, 2024-12-05:
Can't reproduce with yesterdays code in main (can't compile right now to test with latest latest). But this seems very weird... A few things have changed in draftutils/params.py a few days ago, but none should have produced that... @Roy-043 might have an idea?

furgo16, 2024-12-05:
I can also reproduce:

- latest daily snap on Linux, built from last night's code
- It takes a few minutes to load the example
- It does not crash, but when finally loaded, it outputs this on the Report panel

<details> <summary> Python traceback </summary>

```python-traceback
12:26:34  pyException: Traceback (most recent call last):
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 520, in onChanged
    self.colorize(vobj.Object)
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 533, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
  File "/snap/freecad/1216/usr/Mod/Draft/draftutils/params.py", line 646, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
<class 'RecursionError'>: maximum recursion depth exceeded
12:26:34  pyException: Traceback (most recent call last):
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 520, in onChanged
    self.colorize(vobj.Object)
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 533, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
  File "/snap/freecad/1216/usr/Mod/Draft/draftutils/params.py", line 646, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
<class 'RecursionError'>: maximum recursion depth exceeded
12:26:34  pyException: Traceback (most recent call last):
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 520, in onChanged
    self.colorize(vobj.Object)
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 533, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
  File "/snap/freecad/1216/usr/Mod/Draft/draftutils/params.py", line 646, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
<class 'RecursionError'>: maximum recursion depth exceeded
12:26:34  pyException: Traceback (most recent call last):
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 520, in onChanged
    self.colorize(vobj.Object)
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 533, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
  File "/snap/freecad/1216/usr/Mod/Draft/draftutils/params.py", line 646, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
<class 'RecursionError'>: maximum recursion depth exceeded
12:26:34  pyException: Traceback (most recent call last):
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 520, in onChanged
    self.colorize(vobj.Object)
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCurtainWall.py", line 533, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
  File "/snap/freecad/1216/usr/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
  File "/snap/freecad/1216/usr/Mod/Draft/draftutils/params.py", line 646, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
<class 'RecursionError'>: maximum recursion depth exceeded
```

</details>

<details> <summary> FreeCAD version </summary>

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39454 (Git) Snap 1216
Build type: Release
Branch: main
Hash: a97677ed2f0ee4875321f3043782cf77e3756cc6
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

</details>

Syres916, 2024-12-06:
Investigation findings so far, the last build to open the file file without any additional Report View output is:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39299 (Git)
Build type: Release
Branch: (no branch, bisect started on main)
Hash: 89fcfe95ee08bdd0144d94a75425bc9cac66dbd7
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Behave-Dark-Colors 0.1.1
```

[Commit](https://github.com/FreeCAD/FreeCAD/commit/89049add9eda58763348eec8062573106600f81b) introduces the first batch of log output (not directly causing any errors), if the material dev really needs to know this info then surely the object label should be in each log output line??

Somewhere between [commit](https://github.com/FreeCAD/FreeCAD/commit/89049add9eda58763348eec8062573106600f81b) and [commit](https://github.com/FreeCAD/FreeCAD/commit/3ef47da598e843e1d3286d28431022bcda825dbe) is when the max recursion errors are introduced so it's all on the 25th November 2024.

Syres916, 2024-12-07:
@wwmayer Git Bisect has pinpointed [commit](https://github.com/FreeCAD/FreeCAD/commit/c32628ec0df839bff092283a66a716657c78e7f1) as causing the errors [above](https://github.com/FreeCAD/FreeCAD/issues/18295#issuecomment-2517548870) when opening the BIMExample file. Is the way forward to delete all existing windows in the BIMExample file, recreate them with the new code and raise a PR for the updated file or is there a way to open older files and update the alpha channel _onRestoring_?

closed 2024-12-09 by yorikvanhavre

semhustej, 2024-12-10:
I tested with weekly build. FreeCAD does not crash and opens the file.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

But this is the error message shows up:
```
16:25:39  Migrating Start Workbench to Start command... done.
16:26:09  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
16:26:09  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
16:26:09  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
16:26:09  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
16:26:09  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\BIM\ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\BIM\ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\bin\Lib\site-packages\pivy\coin.py", line 3560, in __getattr__
    raise e
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\bin\Lib\site-packages\pivy\coin.py", line 3552, in __getattr__
    return SoBase.__getattribute__(self, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
16:26:09  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\BIM\ArchSite.py", line 948, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagramPosition")
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\BIM\ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\BIM\ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\bin\Lib\site-packages\pivy\coin.py", line 3560, in __getattr__
    raise e
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\bin\Lib\site-packages\pivy\coin.py", line 3552, in __getattr__
    return SoBase.__getattribute__(self, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
16:26:09  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\BIM\ArchSite.py", line 946, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagram")
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\BIM\ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\BIM\ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
[... 653 lines cut ...]
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
16:26:09  pyException: Traceback (most recent call last):
[... cut ...]
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
16:26:09  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39560-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'

```

furgo16, 2024-12-10:
I'm also getting errors on Linux (see details below), but now the file loads considerably faster than before https://github.com/FreeCAD/FreeCAD/pull/18344 was merged.

<details><summary>Traceback</summary>

```python-traceback
20:54:16  pyException: Traceback (most recent call last):
  File "/snap/freecad/1221/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:54:16  pyException: Traceback (most recent call last):
  File "/snap/freecad/1221/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:54:16  pyException: Traceback (most recent call last):
  File "/snap/freecad/1221/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:54:16  pyException: Traceback (most recent call last):
  File "/snap/freecad/1221/usr/Mod/Draft/draftviewproviders/view_base.py", line 293, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
<class 'AttributeError'>: 'NoneType' object has no attribute 'getLength'
20:54:16  pyException: Traceback (most recent call last):
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
  File "/snap/freecad/1221/usr/lib/python3/dist-packages/pivy/coin.py", line 3991, in __getattr__
    raise e
  File "/snap/freecad/1221/usr/lib/python3/dist-packages/pivy/coin.py", line 3983, in __getattr__
    return SoBase.__getattribute__(self, name)
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
20:54:16  pyException: Traceback (most recent call last):
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 948, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagramPosition")
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
  File "/snap/freecad/1221/usr/lib/python3/dist-packages/pivy/coin.py", line 3991, in __getattr__
    raise e
  File "/snap/freecad/1221/usr/lib/python3/dist-packages/pivy/coin.py", line 3983, in __getattr__
    return SoBase.__getattribute__(self, name)
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
20:54:16  pyException: Traceback (most recent call last):
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 946, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagram")
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
  File "/snap/freecad/1221/usr/lib/python3/dist-packages/pivy/coin.py", line 3991, in __getattr__
    raise e
  File "/snap/freecad/1221/usr/lib/python3/dist-packages/pivy/coin.py", line 3983, in __getattr__
    return SoBase.__getattribute__(self, name)
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
20:54:16  pyException: Traceback (most recent call last):
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 946, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagram")
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
  File "/snap/freecad/1221/usr/lib/python3/dist-packages/pivy/coin.py", line 3991, in __getattr__
    raise e
  File "/snap/freecad/1221/usr/lib/python3/dist-packages/pivy/coin.py", line 3983, in __getattr__
    return SoBase.__getattribute__(self, name)
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
20:54:16  pyException: Traceback (most recent call last):
  File "/snap/freecad/1221/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
[... 538 lines cut ...]
</details>

<details><summary> FreeCAD version</summary>

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39562 (Git) Snap 1221
Build type: Release
Branch: main
Hash: 16ac5f99770e078f97e07d7bcf06654f90846673
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

</details>

P.S. To reporters:
- Please consider using the `<details>` markdown syntax for logs and tracebacks. It avoids excessive scrolling in issues => https://gist.github.com/scmx/eca72d44afee0113ceb0349dd54a84a2
- For Python tracebacks, also please consider using the ` ```python-traceback` syntax for easier to read syntax highlighting of Python tracebacks => https://x.com/jakevdp/status/978111914625589248?

yorikvanhavre, 2024-12-17:
Caused by #18298

PR #18344 (2024-12-09): Arch/Draft: Fix transparency vs. alpha issues
Fix #18295
commit 0607c555 (2024-12-09): Arch/Draft: Fix transparency vs. alpha issues

Fix #18295