FreeCAD/FreeCAD #18296: Trying to transform a sketch activates sketch edit mode instead of transform dialog

maxwxyz, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to transform a Body with a Sketch by selecting the sketch in the 3D view and right click > Transform does not start the transform tool command but enters sketch edit more.

Steps to reproduce:
1. Create a body
2. Create a sketch with geometry
3. Leave sketch edit mode
4. Select the sketch in the 3D view; right click and choose `Transform`
5. Sketch edit mode is launched

Could be related to the various issues with the edit modes.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.1
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.31
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.6
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-08:
@FEA-eng also when trying to transform FEM results pipeline, the task view is opened to edit the results pipeline (not that this would make sense to transform this)

VictorLDunn, 2025-11-03:
Discovered this bug using FC 1.0.2 AppImage for myself a few days ago whilst trying to rotate a sketch a few degrees about the z axis.

Because the sketch was a simple rectangle for a part>slice - making a face from the sketch, then allowed the face to be transformed to the correct orientation. 

Bug also existed in FC 0.21.2,

And still exists in the latest Version: 
1.1.0dev.20251031 (Git shallow) AppImage
Build date: 2025/10/31 03:23:50

If a sketch has no attachment - then being able to Transform would be useful (compared to having to trying to use placement properties !)

If the sketch is attached then selecting Transform could either flag a warning, or open the attachment editor  

So, please keep this issue active for a fix sometime in the future.

PaddleStroke, 2026-01-22:
> Well, the main branch disagrees

Hmm well I don't know what happened, I'm not so fluent in git. On the PR the commit showed clearly Werner's authorship.

Here is the command I used to commit, is this not how it's supposed to be done?

`git commit --author="wwmayer <wmayer@freecad.org>" -m "Sketch: Forward non-default edit mode to base class" -m "Ported from Codeberg commit 66dbf277"`

kadet1090, 2026-01-22:
@3x380V The authorship might be my (githubs?) misdoing - I may have used the "Squash" to merge as github remembers last used option and it might have done something with authorship. I'll try by more careful with not using squash for such commits as I have not assumed it may have an effect on authorship.