FreeCAD/FreeCAD #18300: BIM: FreeCAD reports errors when opening file

semhustej, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- start FreeCAD in safe mode
- switch to BIM workbench
- hit 'Cancel' on the BIM Workbench welcome screen
- open file 
[box2 - Copy.FCStd.zip](https://github.com/user-attachments/files/18012841/box2.-.Copy.FCStd.zip)
- errors show up

I am not able to reproduce outside of the safe mode.

```
19:43:35  Migrating Start Workbench to Start command... done.
19:44:19  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_observer.py", line 85, in slotChangedDocument
    from nativeifc import ifc_tools  # lazy import
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 36, in <module>
    from ifcopenshell import geom
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.<class 'ModuleNotFoundError'>: No module named 'typing_extensions'
19:44:19  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_observer.py", line 85, in slotChangedDocument
    from nativeifc import ifc_tools  # lazy import
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 36, in <module>
    from ifcopenshell import geom
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.<class 'ModuleNotFoundError'>: No module named 'typing_extensions'
19:44:19  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_observer.py", line 85, in slotChangedDocument
    from nativeifc import ifc_tools  # lazy import
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 36, in <module>
    from ifcopenshell import geom
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39416-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
[... 214 lines cut ...]
```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

This was confirmed by another user: https://forum.freecad.org/viewtopic.php?p=795559#p795559

It might be related to this issue: https://github.com/FreeCAD/FreeCAD/issues/18293

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-04:
Can't reproduce on:  
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

Report view shows no output

luzpaz, 2024-12-04:
Wait, I can reproduce if I import as .zip file. But not if I remove rename file by removing .zip extension.

maxwxyz, 2024-12-05:
@yorikvanhavre FYI

yorikvanhavre, 2024-12-06:
I renamed the file to remove the .zip part, and it opens fine for me, both in safe mode and in normal mode...
```
OS: Debian GNU/Linux trixie/sid (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39449 (Git)
Build type: Unknown
Branch: main
Hash: d5be9cdafd792c9f37cc66583bc0e784d5c5d26a
Python 3.12.7, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * dxf-library
  * OpenTheme 2024.9.1
  * parts_library
  * freecad.gears 1.3.0
  * OpenDark 2024.3.13
  * CADExchanger
  * BIM021 2021.12.0
  * ExtremeProDark 2.9.0
```

semhustej, 2024-12-06:
I tried with latest weekly and I can still reproduce.

- it it correct to just rename the file - remove the .zip file extension
- the warning does not show in the small warning window. One has to open the Report view view
- maybe a Windows issue as the other user able to reproduce is also a Windows user?

https://github.com/user-attachments/assets/3493ef1b-229b-4ec9-8055-1cc7bbb1dc4c

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

furgo16, 2025-06-02:
@semhustej with https://github.com/FreeCAD/FreeCAD/pull/21641 merged, does this fix your issue with ifcopenshell and typing_extensions?

closed 2025-06-05 by chennes

semhustej, 2025-06-07:
Tested with latest weekly and the issue is fixed. Thank you!

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```