FreeCAD/FreeCAD #18302: Exception when using Measure and undoing operation

mjack3k, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Application ends up in unusable state when undoing operation when Measure was active. 

Initial part:
[image]

After adding chamfer and measuring the selected edge:
[image]

While Measure is active, Ctrl-Z to undo the chamfer:
[image]

When trying to close the Measure:
[image]

### Full version info

```shell
Basically:
- have a circular hole in your part
- create chamfer on the edge on that hole
- use measure tool on that edge
- use Ctrl-Z to undo the chamfer operation (while Measure is still active)
- tada! it is now stuck in Measure, can't close it, have to restart FreeCAD

It is happening in both Part and Part Design, also for simple straight edges (chamfer, measure, undo, stuck). 
It is possible to close the file and create new one, but the program is already borked. Only restart helps.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-04:
Full About info please. (click that copy to clipboard button and paste into ticket)

luzpaz, 2024-12-04:
Can you also please attach an example file (zip it up and drag/drop it into github textbox)

mjack3k, 2024-12-04:
Example file:
[test1.zip](https://github.com/user-attachments/files/18013901/test1.zip)

It is really pretty minimal, to trigger the problem an operation like chamfer needs to be applied to any edge, then it has to be measured, then undo the operation while still in measure mode.

luzpaz, 2024-12-04:
> Full About info please. (click that copy to clipboard button and paste into ticket)

Thanks, can you also add this?

mjack3k, 2024-12-04:
OS: Void Linux (LXQt/lxqt/xcb)
Architecture: x86_64
Version: 1.0.0.1_voidlinux
Build type: Release
Python 3.12.7, Qt 6.7.2, Coin 4.0.0, Vtk 9.3.1, OCC 7.7.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods:

    Gridfinity 0.6.0

mjack3k, 2025-08-28:
@luzpaz I will try. Life has been busy lately, maybe this weekend.

luzpaz, 2025-09-26:
@mjack3k apologies, didn't see the notification you responded.  
https://github.com/FreeCAD/FreeCAD/releases

pjcreath, 2025-11-12:
This was coincidentally resolved by #25129 and should be working in [today's weekly build](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.11.12).

luzpaz, 2025-11-12:
Closed. Please re-open if you can reproduce.