FreeCAD/FreeCAD #18304: Sketcher: error on trying to delete the first external geometry element after PR #17736

MiqCG, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the improved "Create external projection/intersection geometry" tool in sketcher from [PR #17736](https://github.com/FreeCAD/FreeCAD/pull/17736), you cannot delete **the first** element that you "import". An error appear in the report view at each trying and the line stays. The problem appears with projection and intersection options. 
No problem deleting the other created external geometries.

Bug found by @andesfreedesign 

```
21:43:46  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to delete an external geometry with the given index: 0
21:43:46  Unnamed#Sketch001: App.getDocument('Unnamed').getObject('Sketch001').delExternal(0)
21:43:46  ViewProviderSketch: Not able to delete an external geometry with the given index: 0
```

https://github.com/user-attachments/assets/d5c50df4-aa05-4ebe-9d42-f1f9b4711c58

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-12-04:
Confirmed

maxwxyz, 2024-12-05:
@PaddleStroke FYI

PaddleStroke, 2024-12-05:
Ah that is probably me indeed.

closed 2024-12-05 by maxwxyz