FreeCAD/FreeCAD #18309: Layer: shape appearance not propagated to children

maxwxyz, 2024-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Changing Draft Layer shape appearance, e.g. shape appearance color or transparency does not change the appearance of children / elements inside that layer.

Steps to reproduce:
1. Create a layer
2. Change shape appearance of the layer
3. Move objects in the layer

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-05:
@Roy-043 @davesrocketshop FYI

maxwxyz, 2024-12-05:
[image]
Override children is set to true but has no effect

Roy-043, 2024-12-05:
This is indeed an issue but AFAICT not a regression. The Link array is the cause. It does not have a Shape Appearance (in V1.0) or a Shape Color (in V0.21).

Roy-043, 2024-12-15:
Similar: #16725.

closed 2025-03-09 by maxwxyz