FreeCAD/FreeCAD #18315: TestDraft test fails if the user has most recently turned off fill mode

kpemartin, 2024-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you run FreeCAD, start to draw a draft wire (or perhaps other draft shapes), and in the Task panel uncheck "Filled" this eventually is saved (even if you close the Task without drawing a wire) into the user.cfg file as the `fillmode` value in draft settings.

Having done this, if you then run
```
FreeCADCmd.exe -r TestDraft
```
the `test_downgrade` and `test_upgrade` tests fail.

In the downgrade test, the first created object is a Wire rather than a Face so one fewer downgrade steps exist than the text expects. 

In the upgrade test when Filling is on, a (triangular) Face upgrades to an object called Wire  but whose ShapeType is Face, and further upgrading fails as the test expects, but with Filling off, the Face upgrades to a true Wire again, which can then upgrade (unexpectedly by the test) to another Face. It turns out that with Filling turned off, upgrading can go on forever alternating the object between a Face and a Wire.

Although the environment for automatically running the tests (probably) uses a specific fixed Settings (where fillmode is saved), this is not the case when a developer is checking that a new build passes the tests and there may be non-default Preferences,

It is probably impossible to make the tests completely resistant to unexpected environments, but I feel this particular one should test properly regardless of the saved fillmode setting.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39292 (Git)
Build type: Release
Branch: Issue16849
Hash: 15cfe9d2df30d96d6773c53a1b80ea038d11f3a0
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kpemartin, 2024-12-05:
Looking into this has revealed what I feel is two forms of odd behaviour, and I don't know if I should put in PR's for these:

Upgrading a Face with Fill mode off yields a Wire, which can then be Upgraded to a Face and so on indefinitely. I would see converting a Face into a Wire to be a downgrade, not an upgrade, and that trying to upgrade a Face should yield no further upgrades.

Upgrading a Face with Fill mode on yields an objects called Wirennn with the zigzag "wire" icon in the document tree view, but whose ShapeType is Face, which cannot be further upgraded.

Actually, now that I'm trying this out using the UI to draw objects I seem to get different results than when working from the Python console. In Fill mode a 2-line wire upgrades directly to something called "Wire" but whose ShapeType is Face, and cannot be further upgraded. The Report View shows it as closing the open wire, but nothing about making it into a Face.

The scripted form starts by using Part.makeLine to make two lines with a common endpoint, then upgrades the two lines into a V-shaped wire, the next upgrade yield a similar wire but closed, the next upgrade yields a Face, and then the next upgrade diverges based on fill mode, either yielding an object called Wirennn whose ShapeType is Face, or a Wire.

So perhaps a third odd behaviour is that the V-shaped wire drawn using the Draft (make-)Wire UI is somehow different from the one made from two Part_Features and so Upgrades differently.

I may try to write macros that demonstrate these differences...

Roy-043, 2024-12-06:
Confirmed. The Upgrade command just calls `make_wire` without supplying the `face` argument. Which means `MakeFace` of the created object depends of the current filmode.

Fixing the Draft tests should be done obviously. Changing the Upgrade/Downgrade commands may be tricky as users may rely on the current behavior in their workflow.

kpemartin, 2024-12-06:
So I guess the best plan would be to fix the test, close this issue, and make a new issue for the odd upgrade/downgrade behaviour.

One other oddity I saw was that some upgrades/downgrades (I don't remember which ones) don't honour the delete=False parameter to preserve the items being upgraded/downgraded.

There is always a somewhat philosophical issue when you suspect users may be relying on what turns out to be a bug or otherwise undesirable behaviour, at to whether you fix it sooner to avoid more and more users relying on it, or fix it later as part of a major version update where users may be more receptive of (or at least, less surprised by) the change...

kpemartin, 2024-12-06:
Even for fixing the test, I wonder if the fix should be as local as possible (in test_upgrade and test_downgrade, or perhaps in the Setup method of the test_modifications module) or at a very global level (in handling the -r command-line option, for instance)

I don't actually know where the default .cfg files originate for new users, whether there is such a file in the package or whether all the bits and pieces are made up by the various modules as they find the settings missing.

Also, does running tests using the -r option include the tests one gets by running the separately-built `Tests_run` executable? The output looks quite different and the latter is very definitely a shorter suite of tests.

Roy-043, 2024-12-06:
I would fix the tests as locally as possible. They can also be called individually.

kpemartin, 2024-12-06:
Well, I think I understand how the `fillmode` parameter comes to be. `draftutils.params` has a table which supplies a default (in this case True) for parameters not in the .cfg file. Once you change fillmode (by checking/unchecking the checkbox in a Draft Task panel for certain commands), the value is placed in then in-memory image of the Preferences and eventually written out to the .cfg file.

Perhaps the solution would be for the tests to run starting with an empty .cfg file. Perhaps this should be implied if -r or -t are used, and could be overridden with an explicit -u. Same for the system config file and the -s option.

Actually, I now see there is a `--safe-mode` command-line option which would probably do the job and should just be used when using -r and -t. Now I'm not sure if this just isn't a "user error" on my part for not doing this, or whether -r and -t should imply --safe-mode. The downside of the latter is there is no way of explicitly turning off safe mode if you really want the test(s) to run using a specific non-default config, and I'm not sure if adding -u and -s options would be enough to accomplish this.

Roy-043, 2024-12-06:
These tests can also be run from the GUI. I would just catch the old value of fillmode, change the value, and after the tests reset to the old value. You can do this in the actual test functions or in the setUp and tearDown functions I think.

kpemartin, 2024-12-08:
It isn't always possible to reset this back to its original state, though it can be reset to its original effective value. If things start off with no entry in the cfg file, the value defaults to True, but once the code explicitly sets it True, the cfg file will contain an entry with an explicit True value ("1", actually).

I don't know if this should be a concern, but it is possible to code around it by only resetting the value if it was originally False, rather than the more obvious save-the-value, set-the-value, do-the-test, restore-the-value sequence.

kpemartin, 2024-12-08:
Another example of this problem is that CAMTests.TestPathToolController.TestPathToolController.test01 fails if the units and digits of precision in the Preferences do not result in a sufficiently small epsilon.
My Preferences had been set to Imperial units (inches, etc) but retained the default 3 decimal digits of precision.
The test that fails is in part a round-trip test with all the dimensions of the Tool and PathToolController converting to text and back.
The test failed because a distance of 1.75mm = 0.0688976377952755905511811023622" is chopped to 0.069", which converts back to 1.7526mm, which fails the assertRoughly test whose default allowed error is 0.00001.

It isn't clear to me here whether the remedy should be to run the test with the default of mm/3 digits, or whether the allowed error should be calculated based on current units and digits.

In the former case there should be no allowance for error, the (tool diameter) value of 1.75mm should round-trip exactly because it is exactly representable in the floating-point format, so the check should use assertEqual.

I just tried that (in mm, 3 digits) and the test passed using assertExact. Perhaps this would fail in an environment that does not use IEEE or some other binary or hexadecimal floating-point format.

Something in the back of my mind also worries about the entire concept of storing numeric values as text whose precision is at the whim of user Preferences.

kpemartin, 2024-12-08:
I tried using assertExact and also removing the rounding call on the feed speed. The feed speed is set as "3 in/s" and tested as "76.2 mm/s" which should be exactly equal. The underlying value of 76.2 mm/s is not exactly representable in the floating-point format, but the string "76.2 mm/s" and the 1-decimal-place-rounding of "3 in/s" are both the same not-quite 76.2 value and compare equal.
As it turns out, the parsing of "3 in/s" involves converting the "3" to an exact 3 (floating-point) and multiplying by the conversion factor of 25.4 (mm/inch), which is not exactly representable as a float. Multiplying by 3 amplifies the error in 25.4 and results in a value that is not the floating-point value closest to 76.2. The error could be reduced by having the number parser multiply by 254 and divide by 10, but that might still end up one LSB away from the closest value to 76.2

In any case, the assertExact check works as long as that one rounding call is kept.

kpemartin, 2024-12-08:
Hmmm, it isn't clear how to change the number of decimal places or the units from Python. Just reaching in and changing the Parameter won't work because the application doesn't listen for changes to this Parameter value (e.g. by registering an Observer on the containing group). Instead it is updated from the DlgSettings ui when user Preferences are changed.

Roy-043, 2024-12-09:
It would be better if you create separate issues for the problems mentioned in your recent posts. They are not related to Draft. Maybe a more general discussion should be created?

For the Draft issue I think you can use something like this in `setUp` (untested):
```python
param_grp = App.ParamGet("User parameter:BaseApp/Preferences/Mod/Draft")
self.old_fillmode = None
if "fillmode" in param_grp.GetBools() and not param_grp.GetBool("fillmode"):
    old_fillmode = False
    param_grp.SetBool("fillmode", True)
```

And this in `tearDown` (untested):
```python
if self.old_fillmode is not None:
    param_grp = App.ParamGet("User parameter:BaseApp/Preferences/Mod/Draft")
    param_grp.SetBool("fillmode", self.old_fillmode)
```

kpemartin, 2024-12-10:
Yes, we probably need an Issue for the more general problem of how the environment can affect tests, and suggested ways to  deal with this problem. It would probably remain as an open issue, providing a caveat and possible solutions to people writing tests. Perhaps there is some coder's guidelines that this should be included in.

The problem with TestPathToolController tests can also be spun off into its own issue.

My gut worry about willy-nilly rounding and using text to store numeric values is probably just that, a worry, and does not (yet) deserve its own issue.

kpemartin, 2024-12-13:
The general problem of environment affecting tests has been spun off into its own issue #18474.

PR #18841 (2025-01-03): Draft: Make test_modification tests independent of fillmode
Fixes #18315.

In the issue discussion the more obvious solution (change the MakeFace property of the Draft_Wires) was missed.
closed 2025-01-03 by Roy-043