FreeCAD/FreeCAD #18317: [Windows] C++ exception thrown (boost::filesystem::create_directories:

Kuzma30, 2024-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On startup I see such errors
```
16:27:55  C++ exception thrown (boost::filesystem::create_directories: �������� � ������� [system:5]: "C:\Users\Алекс\AppData\Roaming\FreeCAD\SavedPreferencePacks\Backups", "C:\Users\Алекс")
```
looks like boost dont recognise nonANSI chars in path string

### Full version info

```shell
OS: Windows 11 build 22621
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-06:
@oursland FYI

gurkanctn, 2024-12-31:
The problem is still there at FreeCad 1.0.0.
I cannot use the software with this error, it constantly reports the same error.