FreeCAD/FreeCAD #18318: CAM:  Helix drill operation appears to generate unsafe G-Code

sslupsky, 2024-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When generating g-code using the Helix operation it appears to Safe Height parameter is being ignored in build 39380 and the resulting g-code uses a zero height for a rapid move instead of the safe height.  The generated g-code below illustrates the issue.  Safe height is set to 5mm but there is a rapid move at Z0.000.

```
(Exported by FreeCAD)
(Post Processor: script_module.py, version: 2021-10-21)
(Output Time:2024-12-05 11:09:36.533461)
(Begin preamble)
(Default Configuration)
G90
G21
G17
(Begin operation: Fixture)
G54
(Begin operation: TC: 3.175mm Compression Endmill)
(TC: 3.175mm Compression Endmill)
(Begin toolchange)
T10
M3 S12000.0
G4 S3
(Begin operation: Helix)
(Helix)
(helix cut operation)
G0 Z25.000
G0 Z25.000
G0 X20.655 Y18.335 Z25.000
G0 X20.655 Y18.335 Z0.000
G0 X24.278 Y18.335
G1 Z0.000 F600.000
G3 X17.033 Y18.335 Z-1.700 I-3.623 J0.000 F2000.000
G3 X24.278 Y18.335 Z-3.400 I3.623 J0.000 F2000.000
G3 X17.033 Y18.335 Z-5.100 I-3.623 J0.000 F2000.000
G3 X24.278 Y18.335 Z-6.800 I3.623 J0.000 F2000.000
G3 X17.033 Y18.335 Z-8.500 I-3.623 J0.000 F2000.000
G3 X24.278 Y18.335 Z-10.200 I3.623 J0.000 F2000.000
G3 X17.033 Y18.335 Z-10.200 I-3.623 J0.000 F2000.000
G3 X24.278 Y18.335 Z-10.200 I3.623 J0.000 F2000.000
G0 Z0.000
G0 Z25.000
(Begin postamble)
M5
(RRF Configuration)
(The following should be enabled in)
(the config.g)
(M453)
```

[gcode safe height issue.FCStd.zip](https://github.com/user-attachments/files/18027820/gcode.safe.height.issue.FCStd.zip)

### Full version info

```shell
OS: macOS 14.6.1
Architecture: x86_64
Version: 1.1.0dev.39380 (Git) Conda
Build type: Release
Branch: main
Hash: 797c1180b012a8aee1d67de62880df0f835a731e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-06:
@jbaehr FYI

maxwxyz, 2024-12-06:
@jbaehr thanks

maxwxyz, 2024-12-06:
Duplicate of #8140

closed 2024-12-06 by maxwxyz