FreeCAD/FreeCAD #18319: BIM Workbench - wrong geometry from IFC4

nickger, 2024-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Imported geometry from IFC4 file is wrong: surfaces of revolution are not bounded correctly by the edges. Geometry originally is created in OpenCascade end exported via IfcOpenShell to IFC. Please, check the algorythm of creating faces for this case. The IFC file is attached.
[Project31_db.txt](https://github.com/user-attachments/files/18028105/Project31_db.txt)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-12-05:
I tried changing the .txt file extension to .ifc and opening the file in multiple viewers.

- FreeCAD shows geometry
- Open IFC viewer shows some geometry
- BonsaiBIM shows some geometry
- BIMvision shows no geometry
- Autodesk viewer shows no geometry

This seems like a quite difficult file to interpret. 

FreeCAD:

[image]

BonsaiBIM
[image]

OpenIFC viewer
[image]

BIMVision
[image]

Autodesk
[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

nickger, 2024-12-06:
I am very sory, looks like the reference image from OpenCascade was not attached. 
[image]
This is how it looks. If I convert all surfaces to NURBS and export to IFC, it works and looks correct. But as you see yourself, different viewers show different geometry.

maxwxyz, 2024-12-06:
Does other files have issues as well or is it just this file?

yorikvanhavre, 2024-12-06:
This object is an `#436=IFCADVANCEDBREP(#435);` which is a NURBS object. Although this is indeed supported by IfcOpenShell, it apparently still has hiccups (their own viewer at https://viewer.ifcopenshell.org/ , although it does a better job than the others, also shows some defects). Note that almost no commercial BIM app out there is able to support NURBS objects... Certainly not the big, expensive commercial ones ;)

Since it is ifcopenshell that builds the shape in FreeCAD, this is basically an IfcOpenShell bug. I would suggest opening an issue at https://github.com/IfcOpenShell/IfcOpenShell

maxwxyz, 2024-12-06:
If I understand correctly, fixing this in IfcOpenShell will make it work in FreeCAD in the future?
Closing this then.

closed 2024-12-06 by maxwxyz

yorikvanhavre, 2024-12-09:
Yes, there is anyway nothing we can do on FreeCAD's side, apart from helping transferring the bug report to IfcOpenShell...

luzpaz, 2024-12-09:
Heads up @aothms (https://github.com/FreeCAD/FreeCAD/issues/18319#issuecomment-2523289427)

aothms, 2024-12-09:
These kind of 'snail/donut artefacts' as we used to call them were quite specific to revolutions and OCCT 7.8. We have a workaround though, you could try upgrading. https://github.com/IfcOpenShell/IfcOpenShell/issues/5064

aothms, 2024-12-16:
@nickger True, I think pcurves are still ignored in the source data, also because there is approximately 0 public models available that embed pcurves into the exchange. This also applies to things like topological vertex loops. In that sense shapefix posthoc works nicely, because it works regardless of whether parametric data is present or not.

One person within Bentley (now no longer working there) at some point tried to work towards some sample models for adv brep ifc. If you have models with pcurves, maybe publish them there and see if there is some response form the vendors. https://github.com/buildingsmart-community/Community-Sample-Test-Files/tree/main/IFC%204.0.2.1%20(IFC%204)/NURBS

nickger, 2024-12-16:
For sure they are ignored as they duplicate the information: as soon as one has 3d curve and surfaces, he needs to calculate the projections. OpenCascade does it implicitely only for curves on planes, the rest must be calculated by using ShapeFix_Edge::FixAddPCurve. Important is that the initial data store the orientation of edges in bounding loops as this function creates pcurves always in direction of current edge orientation. Nevertheless, this alone is not enough - after calculating pcurves and adding edges to wires, one need to run ShapeFix_Wire::Perform to be entirely sure that UV coordinates are all correct (especially on periodic surfaces) so the face will be constructed correctly by BRepBuilderAPI_MakeFace.
This is what I meant when I said that math comes first: only when you do everithing right in parameteric spaces you can expect correct topology construction. The tool ShapeFix_Face is too generic, from my experience it works 50/50 and for some reason it does make difference: whether you try to correct already created topological structure or you correct its parts first and construct it from the corrected parts.