FreeCAD/FreeCAD #18331: For shortcuts ending in a number, the number cannot be entered on the numpad

Roy-043, 2024-12-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Switch to the BIM WB.
2. Enter the W,P,1 shortcut, press the 1 key above the alpha keys.
3. Result: Works.
4. Repeat, but now use the 1 key on the numpad.
5. Result: Does not work.

If you create a custom shortcut ending in a number, and enter that number on the numpad when defining the shortcut, the shortcut still will not work with the numpad keys.

If the shortcut is a single number the issue does not occur.

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2025-09-13:
I can no longer reproduce.
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/11 03:30:26
Build type: Release
Branch: main
Hash: 770f769b1f234d0b50bfd65f057c33c151622cd3
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

luzpaz, 2025-11-19:
Updated title to reflect that it's Windows specific