FreeCAD/FreeCAD #18335: Redundant/ Partially Redundant Constraints Not Visible

CoderTom314, 2024-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Constraints in sketcher that are redundant or partially redundant do not appear in the lists of constraints. For example, in photo constraint 20 is partially redundant but not visible in the list. Upon clicking and pressing delete, shows "13:09:15  ViewProviderSketch: Delete: Selection not restricted to one sketch and its subelements"
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Chinese/Hong Kong (zh_HK)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * 3D_Printing_Tools
  * AirPlaneDesign 0.4.1
  * CurvedShapes 1.0.13
  * Curves 0.6.51
  * InventorLoader 1.4.0
  * pcb 6.2023.1
  * Plot 2024.11.26
  * sheetmetal 0.6.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

CoderTom314, 2024-12-08:
I just found out that if you create a new file and empty sketch just create an parabola arc, the moment you try to constraint the base of ellipse to one of the end points of the parabola, the number of constraints will exceed the number of constraints shown in the list. Since this time no external geometry is involved, this shows the problem is just due to the sketcher itself?

luzpaz, 2025-07-21:
* Any relation to #22311 ?