FreeCAD/FreeCAD #18337: Part Design: Redesigned origin/base plane selector is very tiny

depthoffocus, 2024-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new PD base plane/origin representation is very much better in terms of design consistently but appears implausibly small -- are there preferences to control this or to reset before testing, or is this just a work-in-progress?

<img width="1088" alt="Screenshot 2024-12-07 at 09 43 50" src="https://github.com/user-
attachments/assets/82d0fc7f-d008-4259-af1a-895fba1e27e1">

### Full version info

```shell
OS: macOS 14.7.1
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.33
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.51
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-07:
@PaddleStroke @kadet1090 FYI, especially on HiDPI screens too
Cursor for size comparison:
[image]

depthoffocus, 2024-12-07:
> @PaddleStroke @kadet1090 FYI, especially on HiDPI screens too 

I did wonder if high-DPI was making it especially small!

maxwxyz, 2024-12-07:
@kadet1090 maybe a global preference for size (maybe ratio/percentage depending on 3D view size) to use for scaling of
- new core datums
- transform dragger
- axis cross (has separate setting in the preferences)

depthoffocus, 2024-12-07:
> @kadet1090 maybe a global preference for size (maybe ratio/percentage depending on 3D view size) to use for scaling of

Unless I am missing something that will be implemented elsewhere., why not make them scale automatically with the size of the Body they are attached to, like Datum Planes do at the moment? 

A plane indicator doesn't just indicate the orientation of a plane, but also the body relative to those planes, surely. So you kind of need to see these things extend beyond the boundaries of the object, so that you can quickly see which parts of the object fall where in relation to the planes.

I am  not sure how useful the sort of semi-transparent presentation can be if these things aren't guaranteed to reach through the sides of the objects:
 
[image]

maxwxyz, 2024-12-07:
no, that's exactly the issue which this PR fixed - it has constant size depending on the view. You can select them also through bodies.

depthoffocus, 2024-12-07:
> no, that's exactly the issue which this PR fixed - it has constant size depending on the view. You can select them also through bodies.

This feels like a significant step backwards in terms of rapid ease-of-comprehension, then, unless they will be able to be set to be always *larger* than the view.

I guess as long as it is still going to be possible to attach auto-growing datum planes that behave the way they do currently, then this may not be a problem.

maxwxyz, 2024-12-07:
They are all be unified with the new ones, constant size. Nothing grows and shrinks automatically anymore and is out of reach. It works as other CAD applications.
The 'old' way you mention had a lot of issues, e.g. these two, just to mention two of the many these PRs fixes.
- https://github.com/FreeCAD/FreeCAD/issues/16966
- https://github.com/FreeCAD/FreeCAD/issues/12584

depthoffocus, 2024-12-07:
Oh dear :-(

So this issue I have raised is then moot, ultimately, because judging by the pull requests they appear more or less as intended. 

Please feel free to close it.

maxwxyz, 2024-12-07:
@depthoffocus no, the issue is valid. It would be great to have a preference how big the constant scale is, depending on the view

JohnOCFII, 2024-12-07:
Just adding another size comparison screen shot.  Difference on a 16" M1 MacBook Pro between development versions 39315 and 39503.  Likely more appropriate to add to a "HiDPI/Retina display" type issue, if there is an overall ticket for that.

```
OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
  * Curves 0.6.51
```

[image]

[image]

PaddleStroke, 2024-12-08:
Can the @FreeCAD/design-working-group discuss what would be the best default size? Then we can also implement the setting.

pierreporte, 2024-12-08:
I would say around 100 px in normal resolution screen (double that for hi dpi). In the screenshot above it’s around 100 px but in hi dpi so 50 px in normal. Basically, it looks twice too small. You could argue that 150 px or even 200 px are fine. In fact they are but it’s going to be too big if you have a lot of datum.

[image]

maxwxyz, 2024-12-08:
@PaddleStroke 
I'm also fine with the default size on normal DPI screens. HiDPI is too small.
The preference should be coupled with the size/appearance of
- new core datums
- transform dragger
- axis cross (has separate setting in the preferences)

Another feedback from @pierreporte (https://github.com/FreeCAD/FreeCAD/pull/16675#issuecomment-2364330318) was that it could be one color and only show colors on hover. (Light on dark BG, dark on light BG). I agree, that the Z elements (dark blue) are not legible on dark themes currently:
[image]

Color should be similar between those elements, I think @benj5378 has a PR for that: https://github.com/FreeCAD/FreeCAD/pull/16995
[image]

The size setting could be located next to the color settings.

obelisk79, 2024-12-08:
The sizing if this *should* be dynamic. Not a user preference. The old plane selector didn't need this. I run a 1440p display with no dpi scaling and the size of the selector provides and extremely small mouse target. 

Making this a preference requiring user interaction would be a usability regression in my opinion. Most Macs run high resolution displays, this would alienate a very large number of users with unnecessary inconvenience. This should 'just work'.

pierreporte, 2024-12-08:
@obelisk79 So you want the size to be a percentage of screen size, using dpi value to compute the pixel size to match the given percentage? In any case, it should be in the preference. Say we end up on 10 % of screen height, somebody may want to have it at 5 % or 15 %.

obelisk79, 2024-12-08:
In that case a preference seems acceptable, but I would still consider it to be an unnecessary customization. My issue would be with forcing a lot of users to have to modify a preference to achieve reasonable usability which is a very poor practice from a usability perspective. 

I would even go at far as starting that if configurable, it should be considered advanced customization a accessed via parameter editor, not preferences window.

maxwxyz, 2024-12-08:
@obelisk79 IMO it definitely needs a preference, same as the coordinate system has today - a dynamic ratio depending on the 3D view size (default 10%). Depending on how far your screen is away or how good your sight is, a preference is valuable.

obelisk79, 2024-12-08:
Why do you all want this to be so small in the first place? How is small better than the previous large? What is the improvement gained by reducing the size of the plane selector by a significant factor?

maxwxyz, 2024-12-08:
I don't want it smaller, it could also be the same as the transform dragger. When working with multiple objects or assemblies, the LCS all clutter the view quite fast (this issue is solved by the new style). When looking at other CAD applications, they are similarly sized and do not grow with the object size (as it did before).

pierreporte, 2024-12-08:
> When looking at other CAD applications, they are similarly sized and do not grow with the object size (as it did before).

There are application that have automatic sizing (Solidworks, Alibre) and other that have fixed sizing (Catia). Personally I prefer fixed size because I find it easier to work with when there are a lot of datum.

obelisk79, 2024-12-08:
The sizing should be a fixed % relative to the display. It should be large and not obscured by 3d geometry. Constraining this to a default 10% of display will result in a much smaller size compared to the old solution. Why shouldn't it be 30% or 40%? What problems would that pose?

maxwxyz, 2024-12-08:
I think we did not specify the default yet? IMO it should not be bigger as the transform dragger as this already gets sometimes in the way of viewing other objects.

obelisk79, 2024-12-08:
This doesn't need to replace the LCS display. I thought this is only used for selecting planes when creating datums and sketches? What is the scope here? Should we move this to discord to prevent cluttering this issue until we have more consensus?

maxwxyz, 2024-12-08:
The main feature which adds LCS and datum objects to core is not merged yet. Basically it allows to add datums and coordinate systems outside PartDesign bodies and use them for assemblies and other attachments. So too many would clutter the view:
[image]

This way you can quickly see objects orientation and references:
[image]

JohnOCFII, 2024-12-08:
> This doesn't need to replace the LCS display. I thought this is only used for selecting planes when creating datums and sketches? What is the scope here? Should we move this to discord to prevent cluttering this issue until we have more consensus?

I think this gets to the heart of the issue (one of them, anyway).  My use case (and primary concern) is **just** the initial selection of a plane when starting a new sketch (as shown in my screen shots above)!  That is where I think a large and easy to read plane selector makes perfect sense.  For the other example that @maxwxyz shows, I would agree you don't want a large graphic.  Can we separate out the two use cases?

maxwxyz, 2024-12-08:
Maybe in case of a command calling for selecting references, they can be enlarged?
@PaddleStroke you wanted to work on the attachment dialog next so maybe this can be combined.

depthoffocus, 2024-12-08:
> Maybe size can increase automatically when a command is started which asks for a selection input?

That would satisfy my main concerns about this, from the perspective of how the old-style planes help a new or cautious user 1) get past blank page fear and 2) sit and ponder for a moment which would be the best choice for a sketch added to an existing model, which I feel is what @JohnOCFII is talking about. 

Having thought about this and read through these comments I think I understand the broader objective is to be able to have this datum information accessible on screen in routine situations where there are more co-ordinate systems in play, and I understand their application there. (Though I note that the images on this page only demonstrate them at the edges of cubes that have a vertex at the origin, which seems an over-simplified scenario)

I'm just one relative CAD novice whose opinions are louder than their value but I would welcome a way to get back the old style planes as a visualisation/comprehension tool alongside the clipping plane tool (or maybe even merged into it) where I could isolate a part and see the big planes clearly cut through the object. I do switch them back on sometimes just to get this visualisation, which helps, I feel I have seen this being done in Youtube videos occasionally, and I feel that maybe some novice-user value is being lost here, particularly if this will also affect datum planes in the Part Design sense.

PaddleStroke, 2024-12-09:
[image]

I have tried on two different screens, one 14" 3000x2000 with 200% dpi scaling. One 38" 3850x1660 with 125% scaling. And indeed the LCS does not scale with DPI. Hence it looks small on the small screen.

I also agree with @JohnOCFII, for many users the only interaction with LCS is when creating a sketch. And for them the target is small.

And I think @maxwxyz has the correct solution. For the sketch selection we could increase the size of the origin object temporarily. So the solution would be : 

- Make LCS take DPI scaling into account. So that they are not twice smaller if you have 200% scaling.
- Make LCS size bigger when creating a sketch.

PaddleStroke, 2024-12-09:
Also as a side note, you currently already have a parameter to change the size of the LCS : 

```
    ParameterGrp::handle hGrp = App::GetApplication().GetParameterGroupByPath("User parameter:BaseApp/Preferences/View");
    return hGrp->GetFloat("DatumsSize", 25);
```
So if you want to temporarily fix this on your side, you can create this parameter and put it a higher value.

depthoffocus, 2024-12-09:
> So if you want to temporarily fix this on your side, you can create this parameter and put it a higher value.

Great! I understand how to do that so I will give that a try later.

depthoffocus, 2024-12-09:
OK I have experimented with that, and I actually think boosting the size will help a lot in the initial selection stage.

One thing I would observe is that the planes do not seem to indicate that they are clickable targets in that context. I can understand them not being clickable targets in other situations of course.

PaddleStroke, 2024-12-09:
@depthoffocus yes it's an issue, they don't show that they are preselected : https://github.com/FreeCAD/FreeCAD/issues/18385
I will have a PR soon to fix this.

depthoffocus, 2024-12-09:
> @depthoffocus yes it's an issue, they don't show that they are preselected : #18385 I will have a PR soon to fix this.

Great -- they also don't really have mouseover behaviour (cursor changes). 

But I think I get it now, and if you can make them sort of scale up and down for initial sketch plane selection, like 30% of screen size or whatever, perhaps with a separate scale setting for that function, I feel it will be a lot more practical.

PaddleStroke, 2024-12-11:
FYI, we have merged already 2 fixes : 
- Preselection not working
- Not scaling with DPI scaling (ie appear very small on hidpi)

If you can build and test again?
I am now going to work on making the LCS planes bigger temporarily when sketch creation.

depthoffocus, 2024-12-11:
> If you can build and test again? I am now going to work on making the LCS planes bigger temporarily when sketch creation.

I am unable to build at the moment and dependent on the weekly builds (which appear to currently be arriving exactly once per week as promised for macOS-x86_64, rather than twice) but I will check when the next one is available. :-)

PaddleStroke, 2024-12-12:
Yes these were preliminary patches fixing scaling issue on hidpi screen. But the view still had to be reworked. 
Here's the PR I just published to fix this : https://github.com/FreeCAD/FreeCAD/pull/18442

depthoffocus, 2024-12-14:
Yes -- the tool is as @JohnOCFII says -- a little larger. But the mouse hover-over action works well now.

GeneGH, 2024-12-28:
Somehow the gist of the problem has been lost in all the discussion.

The issue is that the selection planes that appear when starting a sketch are too small to be easily used. 

From the user view this is unrelated to Assembly or multiple LCS display. 

Not sure how it is coded, but the user issues are completely separate. The treatment should be separate as well.

JohnOCFII, 2024-12-28:
> Not sure how it is coded, but the user issues are completely separate. The treatment should be separate as well.

Agree 100%

Watching some "new user" videos recently, this initial plane selection **needs** to be easy.  Especially if it is going to change away from the many, many videos, (including some new ones released with 1.0) that show the older style (which, IMO, was very clear and useful).  If it were up to me, to @GeneGH's point, I'd stick with the existing initial plane selection, and use whatever new thing you need for the more complex Assembly or LCS selections further down the road.

closed 2024-12-30 by maxwxyz

JohnOCFII, 2025-01-01:
It's better now -- but still not as useful as the previous version.  Hard to select planes when the model gets highlighted and the planes don't extend past the model.

Previous Plane Selection: 

[image]

Current Plane Selection: 

[image]

maxwxyz, 2025-01-01:
that seems to be the transparency issue on macOS. You should be able to click through the model and select the planes.

pierreporte, 2025-01-02:
Even then, if lighting was better, selected faces would be less bright and the LCSs would be more visible. If they were completely white as I proposed, the contrast would be better (provided the lighting problem is fixed). In Catia, there is no problem (except the size if you aren't used to it) unless you make the part completely white.

maxwxyz, 2025-01-02:
About the planes itself, maybe they could be centered so you know where the origin of the sketch would be. No in the enlarged view they are far off to the side.

pierreporte, 2025-01-02:
Yes, this would be close to the NX style I proposed in the first core datum PR:

[image]

BootsSiR, 2025-01-05:
I'm new to FreeCAD with 1.0 coming from Fusion 360 which I've been using for almost a decade (respect to the FreeCAD team, I'm loving it).  I really liked the large, easy to view planes and really missed them when I moved to a nightly build and they got super small.  I can see a use case for both and personally, I think an ideal solution would be the ability to toggle the original large plane view on and off via a toolbar button and/or keystroke.  In the midst of a design, being able to see how the planes intersect with your model can be very useful in my experience.

quartzar, 2025-02-14:
On the weekly builds on Ubuntu, the planes are absolutely miniscule for me. Ridiculously so:

[image]

This is on a 4K monitor if that's helpful.

I had read a workaround for setting `LocalCoordinateSystemSize` to something like `5.0`, but this seems to have no effect.  If anybody has a suggestion/workaround for this, that would be extremely useful. I was very excited to start learning FreeCAD finally, and immediately have hit a big issue. Judging by this first being raised in December and having read that it was apparently fixed (see [here](https://forum.freecad.org/viewtopic.php?t=92781)), I'm wondering if this is perhaps a regression?

I also agree that the previous planes were significantly nicer to use - it should be an option to be able to switch between the two.

quartzar, 2025-02-14:
Just noticed this is marked as "closed" - am I completely missing something, or should I open a new issue?

furgo16, 2025-02-15:
@quartzar It would be better to open a new issue with steps to reproduce and your system info indeed.

maxwxyz, 2025-02-15:
https://github.com/FreeCAD/FreeCAD/issues/19605