FreeCAD/FreeCAD #18338: Sketcher: adding a face as external geometry can result in duplicated edges

depthoffocus, 2024-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Apologies if I have missed a report of this. 

In Sketcher, if you try to add an external geometry edge _after_ you've already added the face it is on, it correctly won't duplicate the edge.

But if you add the face after you add some of the edges, it will succeed and you will end up with confusingly-duplicated edges. It's a little too easy to do this by mistake, and it results in a sketch with hidden duplicate edges.

It seems to me as an easily-confused end user that this should either:

a) succeed: that is, silently either remove the "redundant" edges, or add the remaining edges of the face as edges instead of a reference to a face (if I understand this right), and notify the user either way, or...

b) fail: refuse to add the face because some of its edges are already in the sketch.

https://github.com/user-attachments/assets/6d62fcea-9a79-4157-a400-fea276526f78

### Full version info

```shell
OS: macOS 14.7.1
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.33
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.51
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-07:
@PaddleStroke FYI

PaddleStroke, 2024-12-13:
I'm actually surprised that it won't let you add the edge after the face.

I have been thinking a bit on this face problem. Perhaps we should add all the edges independently rather than adding the face.
Or more precisely, we should add all the edges AND the face, but the face reference should contain only the outline. I think this is possible.

maxwxyz, 2024-12-13:
@PaddleStroke why not just add all edges that form the face? Would that break because of TNP if the number / order of edges changes and then break the sketch? What is the benefit of keeping face outline with all individual edges?

PaddleStroke, 2024-12-13:
It's the outlines of non-planar shapes. If you add only the edges then you can't have those obviously. That's why I added the face as ref initially.
Also for the intersection, adding the edges does not help much.
Also for planar faces that are normal to the sketch we are adding a single line, not all the edges.

maxwxyz, 2024-12-13:
true!