FreeCAD/FreeCAD #18339: It is not possible to run CalculiX with 1 CPU.

mac-the-bike, 2024-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Specifying the number of CPUs:
In: Edit->Preferences->FEM->CalculiX and set the number of CPUs to be used

After some digging I found that: src/Mod/Fem/femtaskpanels/task_solver_ccxtools.py
was controlling the number of CPUs to be used.

            # Set up for multi-threading. Note: same functionality as ccx_tools.py/start_ccx()
            ccx_prefs = FreeCAD.ParamGet(self.PREFS_PATH)
            env = QtCore.QProcessEnvironment.systemEnvironment()
            num_cpu_pref = ccx_prefs.GetInt("AnalysisNumCPUs", 1)
            if num_cpu_pref > 1:  
                env.insert("OMP_NUM_THREADS", str(num_cpu_pref))
            else:
                cpu_count = os.cpu_count()
                if cpu_count is not None and cpu_count > 1:
                    env.insert("OMP_NUM_THREADS", str(cpu_count))
            self.Calculix.setProcessEnvironment(env)

If my reading of the code is correct, it can be seen that when the number of CPUs is set to 1, 'num_cpu_pref', OMP_NUM_THREADS is, in my case, set to 12.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (GNOME-Flashback:GNOME/gnome-flashback-metacity)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: Unknown
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

FEM

### Anything else?

See:

[https://forum.freecad.org/viewtopic.php?t=92426](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-07:
@FEA-eng FYI

mac-the-bike, 2024-12-07:
How do I move this into the FEM subproject?

FEA-eng, 2024-12-07:
Do you mean my FEM Roadmap? I just added it there.

mac-the-bike, 2024-12-12:
That's quick work, thanks, but...

Could you add something to the "Number of CPUs", like:

"Number of CPUs to use:  0 implies use all CPUs

PR #18400 (2024-12-13): Fem: Make it possible to run CalculiX with one core - fixes #18339
Fixes #18339
Set preference value to zero to automatically use the maximum number of available cores.

@FEA-eng
commit c02aa87a (2024-12-13): Fem: Make it possible to run CalculiX with one core - fixes #18339
closed 2024-12-13 by chennes