FreeCAD/FreeCAD #18341: Freecad 1.0 is unsusable at my Linux PC

OliverWieland, 2024-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I start FreeCAD 1.0 (both AppImage and Snap Image), it takes more than 20 seconds to show the start screen.
After I selected  "New File",  nothing happens anymore and all I can do is kill the process.
I use FreeCAD since years and didn't have any such problems with any prior versions on this PC

### Full version info

```shell
OS: Ubuntu 24.10 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-07:
Can not confirm

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.0; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39416 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.0; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.0; Silk 0.1.5; Curves 0.6.51
```
</details>

tarman3, 2024-12-07:
Try to rename folders
`/home/user/.cache/FreeCAD`
`/home/user/.config/FreeCAD`
`/home/user/.local/share/FreeCAD`

maxwxyz, 2024-12-07:
Try restarting FreeCAD in safe mode

OliverWieland, 2024-12-08:
Deleting / renaming the folders, as @tarman3 wrote, didn't have any effects due to my problem.
When starting in safe mode, it starts after 20 secs, but when I try to open a new project, the program freezes .

May this message  be the issue?
`(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""`

tarman3, 2024-12-08:
I have the same error message and FreeCAD is working without problem

OliverWieland, 2024-12-08:
Maybe a graphic engine problem?
`$ lspci | grep VGA
0a:00.0 VGA compatible controller: Intel Corporation DG2 [Arc A770] (rev 08)`

tarman3, 2024-12-08:
Maybe this problem starting after update Ubuntu to 24.10

OliverWieland, 2024-12-08:
With FreeCAD 0.21 I had no issues on (k)ubuntu 24.10

OliverWieland, 2024-12-08:
I built the releases/FreeCAD-1.0 branch and it runs without issues as well.

bassamanator, 2024-12-09:
For me `FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage` will not even launch. I also went ahead and deleted the folders mentioned by @tarman3.

```
Operating System: EndeavourOS 
KDE Plasma Version: 6.2.4
KDE Frameworks Version: 6.8.0
Qt Version: 6.8.1
Kernel Version: 6.12.1-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 2700X Eight-Core Processor
Memory: 31.3 GiB of RAM
Graphics Processor: AMD Radeon RX 580 Series
Manufacturer: Micro-Star International Co., Ltd.
Product Name: MS-7B87
System Version: 1.0
```

X-Ryl669, 2024-12-12:
Add the same issue and it was due to AppImageLauncher (yes, it's unrelated). Delete AppImageLauncher package (you don't need it anyway anymore) and retry.

luzpaz, 2024-12-12:
Related to:  
* https://github.com/FreeCAD/FreeCAD-Bundle/issues/331

Closing

closed 2024-12-12 by luzpaz

OliverWieland, 2024-12-12:
Sorry if this is a silly question, but what is AppImageLauncher?
I searched on my PC, but I did not find any file with this name.
Also I have the same issue with the Snap Image, so I'm not sure if my problem is related to this

X-Ryl669, 2024-12-12:
I was answering to @bassamanator's comment. The AppImage image didn't start because the AppImageLauncher was interfering with it, that's why it doesn't launch. 

Your issue is likely different. I had the same issue as you as well when 1.0.0 was released (the weekly build before were working for me). I've removed all the freecad related folders like @tarman3 said (if I remember correctly, I had more of them but since I've deleted them, I can't tell you anymore). I had an issue with an addon that was incompatible with the new FreeCAD code and my graphic card driver that was outdated and didn't launch correctly.

I updated my drivers and kernel (at the same time) and it worked.

IIUC, The AppImage embeds its libraries (like the glibc) but links with the system's OpenGL's libraries (indirectly, via some drivers defined in ICD files on your system). However, since those drivers uses the system's library and you can't have 2 version of the same library loaded in the executable, it fails (best case) or (worst case) the graphical drivers somehow works but erratically or slowly. 

When it happened for me, the AppImage was using an outdated GLibC (let's say version 2.30), but when FreeCAD tried to load the OpenGL driver, the driver on my system was built with GLibC version 2.31. Since (the old) GLibc was already loaded, the driver wasn't loading (missing symbols in GLibC 2.31) and the default, software CPU rendering loaded instead. FreeCAD was painfully slow. That was solved later on by the AppImage rebuilt to use a newer GLibC, so the opposite happened (AppImage with GLibC 2.32, driver with 2.31, all symbols found => OK). But it's very fragile, as soon as my system's GLibC is updated, it almost always breaks my AppImages until they sync up.

In that case, when you build FreeCAD from source, it'll work on your system, since everything will use the same libraries (yours). 

If I remember correctly, you can run FreeCAD in a console with verbose debugging and it'll show the issues with graphical context.
If it doesn't show anything, try to run:
```
$ gdb /path/to/FreeCAD.AppImage 
[...] 
(gdb) run
Hit Ctrl + C once FreeCAD window is displayed
[...]
(gdb) info symbol GL<press tab here for gdb to complete the symbol>
LIBC_2.10         GLIBC_2.13         GLIBC_2.16         GLIBC_2.2.5        GLIBC_2.23         GLIBC_2.26         GLIBC_2.29         GLIBC_2.3.3        GLIBC_2.31         GLIBC_2.34         GLIBC_2.38         GLIBC_2.5          GLIBC_2.8          GLIBC_PRIVATE      GLOBPAT_NONE
GLIBC_2.11         GLIBC_2.14         GLIBC_2.17         GLIBC_2.2.6        GLIBC_2.24         GLIBC_2.27         GLIBC_2.3          GLIBC_2.3.4        GLIBC_2.32         GLIBC_2.35         GLIBC_2.39         GLIBC_2.6          GLIBC_2.9          GLOBPAT_BACKSLASH  GLOBPAT_SPECIAL
GLIBC_2.12         GLIBC_2.15         GLIBC_2.18         GLIBC_2.22         GLIBC_2.25         GLIBC_2.28         GLIBC_2.3.2        GLIBC_2.30         GLIBC_2.33         GLIBC_2.36         GLIBC_2.4          GLIBC_2.7          GLIBC_ABI_DT_RELR  GLOBPAT_BRACKET    global_offset
(gdb) 
Hit Ctrl + D to exit GDB
```
The highest symbol you'll see is the version the AppImage was linked with (in my case it's GLIBC_2.39)

Then check your driver's required GLIBC version:

```
$ objdump -tT /usr/lib/libvulkan_intel.so | grep GLI
```
Check if the highest value match the one found in the AppImage. It should be less than or equal to the GLibC version, not higher.
Please note that the intel library above match my driver, you'll probably need to adapt to your system.

OliverWieland, 2024-12-12:
Thank you for the detailed explanation!
I don't know why, but now the AppImage works without any issues without changing anything (besides a few package upgrades in the last few days).
So I keep my fingers crossed that the issue is gone...
Oli

bassamanator, 2024-12-13:
Thanks for the detailed explanation @X-Ryl669. I thought I should just build myself, however, finally after getting the build started, it fails at the 76% mark, and I can't even tell what went wrong.

In any case, I uninstalled appimagelauncher, updated the system, rebooted and now the freecad appimage is working, so I'm good to go.