FreeCAD/FreeCAD #18343: Add command line flags for additional macro search paths

jmickelin, 2024-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

#17660 added `--module-path` command line arguments, to allow systems like Nix to augment the add-on search paths with ones supplied on the command line. This way, add-ons can be installed system-wide, but without modifying the predefined `Mod` directory (which would be non-trivial in Nix).

However, the logic for macros searches a different set of directories -- one for User Macros and one for System Macros. Adding similar command-line flags for augmenting each of these would be a welcome followup to the previous work.

### Full version info

```shell
OS: NixOS 24.11 (Vicuna) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Manipulator 1.5.7
  * symbols_library
  * Help 1.0.3 (Disabled)
  * ExplodedAssembly
  * Assembly4 0.50.15
  * workfeature
  * MeshRemodel 1.10.34
  * Gridfinity 0.6.0
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.51
  * fasteners 0.5.32
  * Pyramids-and-Polyhedrons
  * OpenTheme 2024.5.16
  * CurvedShapes 1.0.13
  * workfeature-macro
```

### Subproject(s) affected?

None

### Anything else?

The need for this feature was previously discussed in relation to Nix in NixOS/nixpkgs#362835 and https://github.com/NixOS/nixpkgs/pull/347776#issuecomment-2525086493.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-11:
cc @ein-shved any interest in taking this on ?

luzpaz, 2024-12-11:
cc @galou just a heads up

PR #18525 (2025-01-06): Macro: allow to specify extra system paths for macro
This introduce new option `-E [ --macro-path]` to specify extra system paths of macros. The macro found in this paths will appear in `Macros` dialog at `System macros` tab.

Will fix #18343
closed 2025-01-06 by yorikvanhavre