FreeCAD/FreeCAD #18347: Saving inside sketcher hides any underlying model (Linux, Wayland)

zorael, 2024-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you're in the sketcher on a sketch where you can see an underlying model, such as when you're placing the sketch on a face of a solid, saving the project to file hides the model and only leaves the sketch. It remains hidden until you exit the sketcher.

Some quick troubleshooting suggests it's limited to when you're editing a *previous* sketch in the tree of operations. I don't know the terminology, but this doesn't seem to happen when saving when at the front of the operations history, such as if you are creating a new sketch.

### Full version info

```shell
OS: EndeavourOS (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * A2plus 0.4.68
  * Curves 0.6.51
  * fasteners 0.5.29
  * Gridfinity 0.6.0
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/user-attachments/assets/d6c0e506-40c7-41d4-b24d-b63c22bef090

If it's not apparent, I hit Ctrl+S at 00:07, 00:21 and 00:31.

I have Software OpenGL disabled and OpenGL VBO enabled.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-11-15:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.20251111 (Git shallow)
Build date: 2025/11/11 16:45:10
Build type: Release
Branch: main
Hash: 4fbb389ce1295e04a5ad7f3a63f68b63b4f4036c
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2025-12-06:
@chennes any idea why this occurs ?

closed 2026-09-08 by maxwxyz