FreeCAD/FreeCAD #18350: B-Splines inside a Sketch on Surface doesn't show control points

duub, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to draw a b-spline inside a Sketch on surface but I'm not able to edit the control points. If I create a b-spline outside this sketch the control points are visible and work properly
Screenshot of the diferences, first is inside the sketch on surface and then outside:
[image]
[image]

example file:
[pintura.zip](https://github.com/user-attachments/files/18051672/pintura.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Spanish/Spain (es_ES) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.29
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-08:
@AjinkyaDahale FYI

AjinkyaDahale, 2025-01-04:
I'm not sure why it doesn't show control points in the beginning, but IIRC there was an issue with this in the curves workbench (an addon). The solution might have been to remove the control points. I was able to select the B-spline, then click "show/hide internal geometries" to get the control points, as well as to edit them.

I can't provide further information without knowing what's tool are you using to create the sketch on the surface. Something from the curves workbench?

duub, 2025-01-10:
The visibility of the different items has to be enabled/disabled separately when you work in this different context, didn't occur to me this kind of behaviour, I discovered it by chance.

duub, 2025-01-10:
then it works... but not sure if it's still a bug

closed 2025-01-10 by duub

AjinkyaDahale, 2025-01-11:
What do you mean by "different items" here? Internal geometries of different curves in the same sketch? Different internal geometries of the same sketch? Internal geometries in different sketches?

duub, 2025-01-16:
by items I mean this:
[image]
sorry for not being precise.