FreeCAD/FreeCAD #18355: TechDraw: View boundary, view label, and cosmetic points look like part of the view

pierreporte, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The bounding box of the view, the cosmetic points at the intersection of lines, and the label of the view are the same color of the geometry. The label also have the same font as dimensions. While they are not printed in the exported PDF, it can appear to the user that they are part of the drawing and not only helpers.

I think there should be visible difference between the actual drawings and helpers that are displayed to help users to make drawings. I would recommend:

* changing the color of bounding box;
* changing the default color of cosmetic points;
* changing the view label color;
* changing the view label font to standard app font, and keep its screen size constant;
* putting the view label outside the bounding box.

The 3 colors could be the same, e.g. blue.

[image]

@FreeCAD/design-working-group @WandererFan 

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * sheetmetal 0.5.4
```

### Subproject(s) affected?

TechDraw

### Anything else?

Catia example:

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-12-21:
```
    changing the color of bounding box;
    changing the default color of cosmetic points;
    changing the view label color;
```
These are pretty easy. Most of the work is changing the preference dialogs.

```
    changing the view label font to standard app font, and keep its screen size constant;
    putting the view label outside the bounding box.    
```
Not sure about these. Changing the font is easy enough.  Don't know how to keep the text small when zoomed without some digging.  Putting the label outside the border needs some thought too as it may affect the boundingRectangle calculation.

pierreporte, 2025-01-19:
The more I think about this, the more I think that the current view name label could be entirely removed, and replaces by an optional and customizable label that would be shown in the final document. A lot of CAD programs don’t display the view name unless the user specifically wants it to be present in the final document. It is always available in the tree view. This way, it would make fixing the issue straightforward.

The new persistent label could be fixed to a point that depend from the view content, so it moves automatically as needed. There could be options to put it in the 8 common position (bottom center, top right, etc.), and an options to move it manually.

@FreeCAD/design-working-group @FreeCAD/cad-advisory-group What do you think?

@WandererFan Is it easier to transform the existing label into an actual part of the drawing?

WandererFan, 2025-01-25:
We have the caption property that might meet your persistent label requirement if we added some positioning options.

As for the frame, etc looking like part of the drawing, what about this: 

[image]

pierreporte, 2025-01-25:
@WandererFan It’s a good starting point.

closed 2025-08-11 by chennes

pierreporte, 2025-08-11:
I don’t know if this issue should be closed by #22869. There is still the problem of points and, to a lesser extent, the label of the view that is still at a constant paper size instead of constant screen size.

@ryankembrey do you plan to do a follow-up PR for this?

pierreporte, 2025-08-11:
Thanks!

@maxwxyz Do you prefer reopening or that I create a new one?

maxwxyz, 2025-08-11:
I think dedicated PRs for each WB would be best

pierreporte, 2025-08-11:
@maxwxyz I don’t understand your comment. This is still all about TechDraw.

WandererFan, 2025-08-12:
> the cosmetic points at the intersection of lines

You can set the vertex colour on the colours tab of the TD preferences.   I usually have my preferences set so the vertices are a different colour than the lines.

> the label of the view that is still at a constant paper size instead of constant screen size.

After the recent changes by @ryankembrey, the label only appears on mouse over.  Is this still relevant?

pierreporte, 2025-08-12:
@WandererFan The idea is to change the default color of points in the preferences. I do think that not having constant screen size, and to a lesser extent the same font as dimensions and annotation, is still sub-optimal. I’ll edit the original comment so that @maxwxyz can reopen it (I can’t).

pierreporte, 2025-08-12:
@maxwxyz Edit done

pierreporte, 2026-09-08:
This issue is not fully resolved by PR #22869 because view labels are still constant paper size and the default point color is still black.