FreeCAD/FreeCAD #18356: Sketche External reference line defaults to geometry, but vertex does not

macdroid53, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new feature that allows Sketcher external reference to be used as geometry and is visible w
hen the sketch is closed shows lines that have been referenced but not vertexes.
In the example file, The HeaderBoxFitting and SprinklerHeadLocations have vertex references to Layout Sketch. But, to be seen a sketch vertex needs to be added.
[IrrigationLayout.FCStd.ZIP](https://github.com/user-attachments/files/18052628/IrrigationLayout.FCStd.ZIP)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39509 (Git)
Build type: Release
Branch: makepkg
Hash: 46f561e2c585789e5302ba285422b60008ecd5e3
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * DynamicData 2.70.0
  * lattice2 1.0.0
  * sheetmetal 0.6.1
  * Manipulator 1.5.7
  * kicadStepUpMod 11.4.4
  * fasteners 0.5.33
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.3.0
  * ExtremeProDark 2.9.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

macdroid53, 2024-12-08:
It's more than that. Even toggled the reference vertex does not show when the sketch is closed.

Shkolik, 2025-02-26:
> It's more than that. Even toggled the reference vertex does not show when the sketch is closed.

It's even more than that. If you have external line and external point as defining geometry and then decided to make both construction, after closing sketch external line will remain defining. After that happens you'll nit be able to make any external geo construction - the only way to fix that is removing external point from a sketch. Very weird issue - couldn't find anything wrong in code at a first glance - most likely issue inside rendering code somewhere.

Shkolik, 2025-02-26:
Did a bit debugging:

for some reason in ViewProviderPartExt::updateVisual() if sketch contains only one defining external point Bnd_Box calculation fails. We need it to calculate deflection, but for point it always be 0, so I tried skip this calculation, and it worked - point is seeing in 3d view, but problem I described above still there, so I assume nothing is wrong with updateVisual, it's actually something with geometry itself, since Bnd_Box calculation is working fine for single normal point (not external).

commit 133cf6de (2025-03-05): Merge pull request #19889 from Shkolik/sketcher_external_point

fix #18356 Sketcher: External reference line defaults to geometry, but vertex does not
commit 27dd1417 (2025-03-05): fix #18356 Sketcher: External reference line defaults to geometry, but vertex does not
closed 2025-03-05 by chennes