FreeCAD/FreeCAD #18357: Ubuntu 24.04: OCCT lib dependency issue when having KiCAD installed in parallel

s-hemer, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

KiCAD (still) ship their own Opencascade libs not following the Y2038 naming scheme with the "t64" at the end as the Debian packages do. This reults in an unresolvable dependency issue (at occt-misc) when one wants to install both KiCAD and FreeCAD. I will try to file a bug over at KiCAD as well.

I know, one may argue "it's KiCADs bad, they should fix it etc pp" but I think the easiest fix would be to allow also the KiCAD OCCT libraries. 
Using whatever, flatpak, appimage, .. is also not an option.

I solved it by manually manipulating ```libfreecad-python3-0.20_0.21.2+dfsg1~202407140123~ubuntu24.04.1_amd64.deb```'s control file by adding ```|libocct-xxx-7.6``` to all the OCCT lib dependencies.
So far, I could not see any issues.

### Full version info

```shell
Version 2:0.21.2+dfsg1~202407140123~ubuntu24.04.1 from https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-stable/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

s-hemer, 2024-12-08:
Related KiCAD issue: https://gitlab.com/kicad/packaging/kicad-ubuntu-builder/kicad-daily-package/-/issues/54

maxwxyz, 2024-12-09:
@s-hemer is this also an issue with the stable 1.0 version?

s-hemer, 2024-12-09:
There is no Debian package for 1.0 yet?! There is also the tag ```Packaging/building``` that may be added.

luzpaz, 2024-12-09:
I could of swore there was an open ticket about creating a communication channel open between FreeCAD and KiCAD devs to invite both projects to keep their version support of OCCT similar

s-hemer, 2024-12-10:
> I could of swore there was an open ticket about creating a communication channel open between FreeCAD and KiCAD devs to invite both projects to keep their version support of OCCT similar

Sounds somehow familiar, yes, but that was the time when both build their own Opencascade libs (as their where none in Debian/ubuntu?) but now in 24.04 FreeCAD uses the the distributed ones again. Version-wise that would be no issue but the naming of the package changed and so the ones shipped with KiCAD do conflict.

@adrianinsaval if you need any help for packaging, I am eager to support/test/.., also for packaging v1.0.

luzpaz, 2024-12-10:
@s-hemer jfyi, packaging repo is at https://github.com/FreeCAD/FreeCAD-Bundle

closed 2025-12-05 by maxwxyz