FreeCAD/FreeCAD #18358: TechDraw: Multi-page drawings are not supported

pierreporte, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

TechDraw can only create standalone page. It means that, to create multi-page drawings, it is necessary to manually create a whole page, complete with template selection, and edit title block fields for numbering. It lacks a Drawing container, that holds page size, template, and common settings for the drawings, of which Page containers would be children. Page numbering would then be automatic, and it would allow to easily create multiple drawings in the same file (e.g. simplified assembly drawing for customers, and assembly instructions for manufacturing).

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * sheetmetal 0.6.1
```

### Subproject(s) affected?

TechDraw

### Anything else?

This is a necessary foundation for the new file format currently studied, but it can exist independently.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-12-29:
@WandererFan What's your opinion on this?

WandererFan, 2024-12-30:
I haven't given much thought to this.  At first glance, multipage drawings appear to need a major structural change.  

The part about a container for drawings is probably less invasive than the multi-page drawings.  Can't see the page size being an attribute of the container since there could be different size drawing pages in a single document.  But maybe the vision is multiple "drawing books" in an FC document.

Re automatic page numbers - I've been told in the past that we need the ability to manually change page numbers and number of pages because the finished document will include pages that are not drawings (so TD doesn't know about them).

pierreporte, 2024-12-30:
> The part about a container for drawings is probably less invasive than the multi-page drawings. Can't see the page size being an attribute of the container since there could be different size drawing pages in a single document. But maybe the vision is multiple "drawing books" in an FC document.

The idea is to have something like this:

* Drawing 1
  * Page 1
  * Page 2
  * Page 3
* Drawing 2
  * Page 1
  * Page 2
  * Page 3

Drawing 1 and Drawing 2 are the containers that would hold template and everything, so the two drawings can have a different size. Current pages would be the children of these containers.

I haven't thought about a container for all the drawings. It's an interesting idea that could be used for other things as well, like CAM or FEM.

> Re automatic page numbers - I've been told in the past that we need the ability to manually change page numbers and number of pages because the finished document will include pages that are not drawings (so TD doesn't know about them).

I wasn't aware of that. I would say that it should be automatic by default, and provide the ability to override. Maybe an offset value could be good as well, to allow users to define the starting number.

tobiasfalk, 2024-12-31:
I think this is installed a way related to: https://github.com/FreeCAD/FreeCAD/issues/13913#issue-2287142442

pierreporte, 2025-01-03:
Closing as duplicate.

closed 2025-01-03 by pierreporte