FreeCAD/FreeCAD #18359: Task view disappears

macdroid53, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the course of normal work, it appears after an error has been reported, the task view refuses to show.
When this happens, if a modal dialog is in process there is no way to close the dialog.
In most cases, when this condition arises, a right click in a toolbar area shows the Task view is checked but the word Tasks is greyed.
The only consistent way to restore correct display of the Task view is to restart FreeCAD.
I have provided no file,because it happens with _any_  file.
I have also not been able to define a procedure to duplicate the condition. But, it happens every time I attempt to work in FreeCAD on the platform shown in the version info. It happens whether the Panel is docked or undocked on another monitor.
I have not seen the condition on Debian/XFCE, but one user did report it on macos.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39509 (Git)
Build type: Release
Branch: makepkg
Hash: 46f561e2c585789e5302ba285422b60008ecd5e3
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * DynamicData 2.70.0
  * lattice2 1.0.0
  * sheetmetal 0.6.1
  * Manipulator 1.5.7
  * kicadStepUpMod 11.4.4
  * fasteners 0.5.33
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.3.0
  * ExtremeProDark 2.9.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

macdroid53, 2024-12-08:
Further use appears that this condition occurs after any initial warning or error message.

Syres916, 2024-12-09:
I'll compile the same commit using Qt6.7.2 and see if it behaves differently.

macdroid53, 2024-12-10:
As I note I have only seen this on Manjaro (two different PC's), FC built from AUR freecad-git. (I think @obelisk79 has seen similar  on his Arch based distro). I have never experienced it on Debian 12 (built from source) or Ubuntu with the weekly appimage.

obelisk79, 2024-12-10:
I haven't had the problem in some time so it's likely a dependency related issue that has since been fixed. Manjaro typically lags behind arch a few weeks ago this isn't surprising to hear.

Syres916, 2024-12-11:
My feedback is, I'm so glad I did a snapshot before upgrading 776 packages. While I couldn't reproduce the _Tasks_ DockWidget disappearing no matter what I tried using Qt 6.8.1 the overall performance was so poor hence going back to the original snapshot.

macdroid53, 2024-12-11:
In 39583 I just did it by closing a sketch, Extrude the sketch, open the sketch again...no Task view. And, the view list shows Task checked but grayed out.
Note, the Property view and Task view were un-docked.

macdroid53, 2024-12-17:
Seen again in:
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39675 (Git)
Build type: Release
Branch: main
Hash: d507cf34530837495596c16c1f15ead4dac4cc2e
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * Curves 0.6.52
  * CurvedShapes 1.0.13
  * DynamicData 2.70.0
  * lattice2 1.0.0
  * sheetmetal 0.6.2
  * Manipulator 1.5.7
  * kicadStepUpMod 11.4.4
  * fasteners 0.5.33
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.3.0
  * ExtremeProDark 2.9.0

Dtays-meanderings, 2024-12-18:
```txt
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39646 (Git) Conda
Build type: Release
Branch: main
Hash: b25b83c1a58bf0ae95cbd502d0ccae27e8127cad
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
```

## Steps to Reproduce:

1. Toggle transparency for all objects (`Shift+F4`).
2. Create a line using the Draft workbench and finish/exit the task.
3. Double-click the Draft line object in the tree. Note that a task dialog has opened.
4. Close the task dialog using the `x` button instead of the `close` button.

### Observations:
- The task panel remains greyed out while the task is active.
- Clicking into the 3D view and pressing `Escape` will clear the task, and the panel is no longer greyed out.
- Alternatively, toggle transparent display with `F4` and close the task.

macdroid53, 2024-12-30:
This is getting worse. But, it appears only in Arch/Manjaro.
Almost any error report and the Task will go into limbo. Currently 39747.

xtemp09, 2024-12-31:
> I have never experienced it on Debian 12 (built from source) or Ubuntu with the weekly appimage.

Sorry to intrude into a well-established conversation, but, @macdroid53, this is a clearly problem with _Qt_. Does _AppImage_ work as expected?

macdroid53, 2024-12-31:
> > I have never experienced it on Debian 12 (built from source) or Ubuntu with the weekly appimage.
> 
> Sorry to intrude into a well-established conversation, but, @macdroid53, this is a clearly problem with _Qt_. Does _AppImage_ work as expected?

I don't think it was ever in question that it is a Qt6 issue. The AUR package is freecad-git.

The appimage does work as expected.

xtemp09, 2024-12-31:
I think that this issue and the issue #16640 are strongly related or even the same.

@macdroid53, if you change stylesheets, does the task view show up?

macdroid53, 2024-12-31:
I take that back. The appimage suffers the same task view zombie. It does correct the '&' in panel titles.

macdroid53, 2025-01-04:
As I use this on Manjaro I try to track and possibly find a repeatable sequence to produce the problem. So far no luck.
But, I have observed that the panel tiles with "&" are always there for the Report and Python panels. And, ampersands don't appear in the Property and Task view until an error is reported. The other common thing is the Task view going zombie appears to have something to do with the Task view being hidden after an error.
While it does happen when the panels are docked, this is most prevalent when the Task and Property are separate and and undocked on a separate monitor.

NohWayJose, 2025-01-11:
Just encountered this on:
OS: openSUSE Tumbleweed (KDE/plasmawayland/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3DfindIT 1.2.0
  * Assembly3 0.12.2
  * fasteners 0.5.15
  * frame 0.1.1
  * Help 1.0.3
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * POV-Ray-Rendering
  * Render 2024.1.23
  * sheetmetal 0.4.10
  * Silk 0.1.5
  * ThreadProfile 1.89.0
  * workfeature
  * PieMenu 1.2.7
  * ExplodedAssembly

NohWayJose, 2025-01-11:
just for clarity, I drag-undocked the task panel (with a view to having the main view on one screen and the tools on the other. Not sure how I disappeared the panel (now a window) but it didn't appear in my KDE task panel and only a restart of FreeCAD brought it back

macdroid53, 2025-01-24:
For others who may stumble on the Task view zombie issue: Thanks to help from @hyarion in the Discord the following Python seems to recover from the zombie state.

`Gui.runCommand('Std_Edit',0)`

I have added it to a global toolbar and it has worked every time.

juni3227, 2025-08-27:
I am having that bug too. 
Only thing I did was creating a new cube while the placement UI is open. ... I really have no Idea why this happenned. Is there any logs that I can send instead?

juni3227, 2025-08-27:
https://github.com/user-attachments/assets/39adb38d-4c9a-46be-b577-801637e141af

xtemp09, 2025-08-27:
@juni3227, you should provide About info every time you report a bug.

It is clear from the video that you are using _1.0.0_. Is the problem reproducible with the newer version and with a weekly bundle?

danicc097, 2025-10-20:
One workaround is to toggle overlay for all. In my case it restored the tasks panel which didn't reappear after restarting. 

OS: EndeavourOS (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.2.39319 (Git)
Build type: Release
Branch: makepkg
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.13.7, Qt 6.10.0, Coin 4.0.5, Vtk 9.5.2, OCC 7.9.1

[image]

luzpaz, 2025-12-16:
Is this an Overlay bug ?