FreeCAD/FreeCAD #18360: Gears Workbench + Sketch -> Sketch: No planar face in AttachEngine3D::calculateAttachedPlacement()!

ChuckInGame, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sketch is going in Error, after changing the gear, the sketched is used on

### Full version info

```shell
OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.33
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Anything else?

[image]
[Screencast From 2024-12-08 16-25-50.webm](https://github.com/user-attachments/assets/d51ca1d7-4fc7-4aef-9744-6d01bfb2a5cc)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-09:
Is this an issue with main or an issue with Gears WB[^1]? 

[^1]: https://github.com/looooo/freecad.gears/issues

ChuckInGame, 2024-12-20:
I Added an issue on the gear workbench,
https://github.com/looooo/freecad.gears/issues/195
my be this is related to this or otherwiese

luzpaz, 2025-05-10:
@ChuckInGame  
Looks like https://github.com/looooo/freecad.gears/issues/195 is resolved. What's left to do here ?

FEA-eng, 2025-09-27:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```