FreeCAD/FreeCAD #18361: Gears workbench + Sketch

ChuckInGame, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

May be possible, that it's the same as in #18360, but i got a different error, so I posted an additional bug, please tell me if i should combine those bugs next time.

As in #18360, i added a sketch to a gear to make a hole, which worked as expected.

After that i changed the gear properties, especial the **beta** and **double_helix** property (I think these are the reasons for the errors)
[image]

Result:

* The sketch is marked with an error flag
+ 2 errors occoured
[image]
* The visualisation of the gear will not update any more:

-> see the property vakze  of double-Helix now  and the gear:

[image]

The screencast to reproduce it, is attached

### Full version info

```shell
OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.33
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Anything else?

[Screencast From 2024-12-08 16-54-09.webm](https://github.com/user-attachments/assets/c63dd1c8-2e2f-4566-85ee-5f391e721c39)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ChuckInGame, 2024-12-08:
And a question,

[image]

On the gears, there is always a purple circle in the bottom left corner of the icon, possibly with an exclamation mark or two dots stacked vertically.

What does this signify, and why is it there?

Is it a warning indicating that something is wrong, or is it just part of the icon itself?
If it’s the latter, I would prefer an icon without this circle.

luzpaz, 2024-12-09:
Needs an example file (zip it up and drag-drop into Github textbox)

cc @looooo

Is this an issue with main or an issue with Gears WB[^1]? 

[^1]: https://github.com/looooo/freecad.gears/issues

ChuckInGame, 2024-12-09:
i attached a sample file in the state of the end of the screen recording
- Error occured
- Visualisation will not update any more, even on restart / reload

looooo, 2024-12-11:
Not sure if it is related, but we have an occ issue with helical extrusion:

ChuckInGame, 2024-12-20:
I Added an issue on the gear workbench, 
https://github.com/looooo/freecad.gears/issues/195
my be this is related to this or otherwiese

FEA-eng, 2025-09-27:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * A2plus 0.4.68
  * freecad.gears 1.3.0
```

luzpaz, 2026-09-20:
@looooo you're assigned to this issue. Is that still valid for you?