FreeCAD/FreeCAD #18364: Assembly: create mid-plane joint

eblanshey, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

One very useful joint to add to the assembly workbench would be a mid-plane joint. There are two variations found in other CAD software:

1. Single-plane: select a single plane in object A, followed by two planes of object B. Object A's plane is centered between the planes of object B. 
2. Double-plane: select two planes in object A, followed by two planes of object B. Object A's two planes are centered between the planes of object B.

Examples: single-plane mid-plane can be used to center the XY plane of an object between two walls of another object, without having to calculate the distance between the two walls, or having to create a datum plane within that object in the correct place. Double-plane is useful for objects that have two walls, such as a box, that need to be positioned between the walls of another object, but where a single center plane does not exist.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * WebTools 1.0.0
  * Help 1.0.3
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-08:
With core datums merged soon, they can be attached between two elements and referenced in assembly

maxwxyz, 2025-01-22:
@PaddleStroke is this relevant anymore? Workaround would be using new core datum plane IMO.

PaddleStroke, 2025-01-22:
Well I don't know. I think it would be best if joint creation could accept multiple selection, notably for this case which seems to be fairly common.