FreeCAD/FreeCAD #18367: PartDesign : Perf : making threads via Helix (additive or substractive) tool is only single core. Poor perfs + final crash

bebenlebricolo, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While making a M10 threaded rod (either trying to cut the threads from a cylinder using negative helix or starting with the thread profile), on long rods process takes forever.
I also had a crash once, after waiting for quite some.

Albeit I can make threads on a small portion of the rod (few millimeters), it will run on a single core at 100% for some time (note that the load is dynamically dispatched over the cpu cores as the threads lives its life).
For a longer rod (my use case is to create a 382 mm M10 threaded rod, ISO coarse standard, 1,5mm pitch), freecad will first freeze, then finally crash.

I've linked both the repro file (fairly straight forward) and the associated log file (which only seems to log initialization data..).
[FreeCAD.log](https://github.com/user-attachments/files/18053447/FreeCAD.log)
[ThreadedRods_Repro.zip](https://github.com/user-attachments/files/18053451/ThreadedRods_Repro.zip)

 # How to reproduce
1. Open the linked file
2. Edit the additive helix
3. Set length to 390 (and I believe anything that's high enough) and press enter.

 

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.64
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-09-14:
Still relevant

FEA-eng, 2025-11-15:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.20251111 (Git shallow)
Build date: 2025/11/11 16:45:10
Build type: Release
Branch: main
Hash: 4fbb389ce1295e04a5ad7f3a63f68b63b4f4036c
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

Lgt2x, 2025-12-06:
Related to https://github.com/FreeCAD/FreeCAD/issues/17113 which was fixed by https://github.com/FreeCAD/FreeCAD/pull/17356

To satisfy the tolerance requirement we impose on the helix, OCCT switches to a more precise (and costly) model when the helix is above a certain size. We can easily demonstrate this using a simple script:

```python
import FreeCAD
doc = FreeCAD.open("ThreadedRods.FCStd")
helix = doc.getObject('AdditiveHelix')
helix.Height = '247 mm'
doc.recompute()
```

takes 1.3s on my PC, and 

```python
helix.Height = '248 mm'
```
takes about 7.9s.

We can improve the situation by changing the hard-coded tolerance value in [TopoShape.cpp](https://github.com/FreeCAD/FreeCAD/blob/6ec9870d5456108ce6a354619e406d4369f7a441/src/Mod/Part/App/TopoShape.cpp#L2501). By setting the added factor to `1e-2` instead of `1e-6`, setting the helix size to 382mm takes about 1.5s, and the result looks visually fine.

[image]

@AIRCAP any input? I don't have enough hindsight to justify the value change