FreeCAD/FreeCAD #18368: FEM: Simple ccx fluid section example not working anymore

FEA-eng, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a simple example of 1D fluid flow with CalculiX that used to work in 0.22 dev. Now I've noticed that it's not working anymore in 1.0 or 1.1.0, likely due to some changes in results handling. The following errors appear:

```
19:09:10  Running the Python command 'FEM_SolverRun' failed:
Traceback (most recent call last):
  File "E:\FreeCAD_weekly-builds-39503-conda-Windows-x86_64-py311\Mod\Fem\femcommands\commands.py", line 1110, in Activated
    run_fem_solver(self.selobj)
  File "E:\FreeCAD_weekly-builds-39503-conda-Windows-x86_64-py311\Mod\Fem\femsolver\run.py", line 110, in run_fem_solver
    fea.run()  # standard, no working dir is given in solver
    ^^^^^^^^^
  File "E:\FreeCAD_weekly-builds-39503-conda-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 638, in run
    self.load_results()
  File "E:\FreeCAD_weekly-builds-39503-conda-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 756, in load_results
    self.load_results_ccxfrd()
  File "E:\FreeCAD_weekly-builds-39503-conda-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 766, in load_results_ccxfrd
    importCcxFrdResults.importFrd(
  File "E:\FreeCAD_weekly-builds-39503-conda-Windows-x86_64-py311\Mod\Fem\feminout\importCcxFrdResults.py", line 70, in importFrd
    m = read_frd_result(filename)
        ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\FreeCAD_weekly-builds-39503-conda-Windows-x86_64-py311\Mod\Fem\feminout\importCcxFrdResults.py", line 668, in read_frd_result
    massflow = float(line[13:25])
               ^^^^^^^^^^^^^^^^^^

could not convert string to float: '-1.#IND0E+00'
```

It's working when I submit it using standalone CalculiX. I can even import the results to FreeCAD then.

What's also strange, the input file is split (*INCLUDE keyword is used) even though I have this option disabled.

File: [no_water_1d_fluid 2.FCStd.zip](https://github.com/user-attachments/files/18053463/no_water_1d_fluid.2.FCStd.zip)

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2024-12-09:
What version of CalculiX are you using?
In my case no exception was thrown, but the .frd file contains  NAN values (which Python converts to a nan float value)

FEA-eng, 2024-12-09:
> What version of CalculiX are you using?

2.22 provided with FreeCAD on Windows

marioalexis84, 2024-12-10:
Could you share the .frd file?

FEA-eng, 2024-12-14:
> Could you share the .frd file?

Sure, I attached it here. I can open it in GraphiX and there are some results in it.

[FEMMeshGmsh.frd.zip](https://github.com/user-attachments/files/18135588/FEMMeshGmsh.frd.zip)

luzpaz, 2024-12-15:
> What version of CalculiX are you using?

@marioalexis84 an aside, should we add calculix to the output of the Help > 'About FreeCAD' > 'Copy to clipboard' info ?

marioalexis84, 2024-12-15:
I think it is not necessary, since it is a very specific tool for Fem. But we can add a button in the solvers task panel to get version information as it is done for meshers.

marioalexis84, 2024-12-15:
@FEA-eng could you try using another CalculiX executable (maybe an older version)? I'd like to know why the .frd I get is not exactly the same as yours...

On the other hand, the input file is split because the `SplitInputFile` property is set to `True` in the Solver object. Modified property values ​​take precedence over user preference values.

FEA-eng, 2024-12-16:
> @FEA-eng could you try using another CalculiX executable (maybe an older version)? I'd like to know why the .frd I get is not exactly the same as yours...

It works with the newest version (2.22) and 2.21 when I use ccx binary from the CalculiX website but not when I choose the one provided with FreeCAD. So it seems to be somewhat similar to #10865.

marioalexis84, 2024-12-22:
> should we add calculix to the output of the Help

@luzpaz  I think you're right about this. We'd even gain space in the task panels if we removed the version button.
But I don't think `about` is the right place. Maybe an option to collect information from third-party libraries for each workbench.

luzpaz, 2024-12-22:
> But I don't think `about` is the right place

I don't see why not. It's informative.

marioalexis84, 2024-12-30:
You are right. There is some work in progress to add specific workbench info to the `About` or `Libraries`?

luzpaz, 2025-01-03:
3rd party Workbenches are already added to About dialog. I'll open a new ticket for this so we don't go off-topic.

maxwxyz, 2025-04-07:
@marioalexis84 I'm getting the same error now on all calculix solves. It runs successfully but fails to load the results because of:
```
20:08:16  loading results failed
20:08:16  could not convert string to float: ' 0,00000E+00'
``` 

This is with the latest weekly and the FEM_example file, also in safe mode.

@oursland @adrianinsaval might this be related to https://github.com/FreeCAD/FreeCAD/issues/10865 with the recent build changes of the weeklies? Can anyone reproduce?

[image]
Also, it wipes all result objects:

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41155 (Git) Conda
Build date: 2025/04/06 04:00:07
Build type: Release
Branch: main
Hash: e537278f1296211edb90bf1c938dfaebd2626f0d
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
```

marioalexis84, 2025-04-07:
What language do you have on your system?
>`Could not convert string to float: ' 0,00000E+00'`

It looks as if the results in the .frd file are written using a comma as a decimal separator.

maxwxyz, 2025-04-08:
@marioalexis84 System is German but FreeCAD in English or German.

marioalexis84, 2025-04-08:
@maxwxyz Could you share one of the .frd files generated by CalculiX in the working directory?

maxwxyz, 2025-04-11:
@marioalexis84 here are the files:

[fcfem.zip](https://github.com/user-attachments/files/19700803/fcfem.zip)

Problem still exists in

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda
Build date: 2025/04/09 18:28:54
Build type: Release
Branch: main
Hash: de124a72b56515a34f8ed627a768b54695698fc8
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.30.1
  * CurvedShapes 1.0.13
  * Curves 0.6.60
  * dodo 1.0.1
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Gridfinity 0.11.6
  * Manipulator 1.5.8
  * OpticsWorkbench 1.1.3
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.22
  * Silk 0.1.5
  * Telemetry 1.0.0

```

marioalexis84, 2025-04-11:
The .frd file generated by CalculiX has a comma as the decimal separator instead of a dot. This shouldn't happen. Even cgx loads the file incorrectly if a dot isn't used as separator.
Is your system language (not FreeCAD) using a comma as the decimal separator? Try modifying the language to use a dot.

maxwxyz, 2025-04-11:
@marioalexis84 yes, as most countries we use a comma as decimal separator. It has worked before though and it's not ideal to manually switch system settings each time I start FreeCAD 😄

marioalexis84, 2025-04-11:
Of course, it was just to test and make sure that's the problem. It seems to be a bug in CalculiX. Can you try an older version of CalculiX?

maxwxyz, 2025-04-11:
How? I did not install CalculiX manually. Changing number format at system level works as workaround.

marioalexis84, 2025-04-11:
http://www.dhondt.de/calculix_2.22_4win.zip
And set the path to ccx executable in Fem->CalculiX preferences.

Probably, this issue is just another variant of https://github.com/FreeCAD/FreeCAD/issues/10865

maxwxyz, 2025-04-11:
@marioalexis84 it works with the provided executable without issues. So is it the same as #10865 ? If so, when did we change the executable whipped with FC?

oursland, 2025-04-11:
A while back I added the standard workbench components that were failing on the weekly builds including:

* pycollada
* openifcshell
* calculix

If the calculix in `conda-forge` is not working, my preference would be to fix it over remove it.  In general my belief is that the all of the default workbenches should work out of the box on an install, without requiring the user to search and install each dependency and match their versions to the version used with FC.

One area that this is an issue, but is not a default WB so I have not addressed it, is the CfdOF workbench that requires a specific version of OpenFOAM installed, which is also not the latest version.

FEA-eng, 2025-08-11:
We talked about this with Mario and we think that it might be caused by the fact that the machine used to prepare Windows builds has comma as a decimal separator. @adrianinsaval Do you know something about it?

Anyway, perhaps the easiest fix would be to provide Windows version of FreeCAD with CalculiX binary from the official website instead of the self-generated one. This should solve other issues with CalculiX binary too (#10865). I hope it's possible to package FreeCAD this way. Here's a link to the newest official build: https://www.dhondt.de/calculix_2.22_4win.zip

maxwxyz, 2025-11-07:
@adrianinsaval ping

stefano2734, 2025-11-17:
Calculix 2.23 available.

adrianinsaval, 2025-12-29:
the patch works but then gives a different error on some cases:
```
could not convert string to float: '   -NAN(IND)'
```
I'll make a PR to sanitize this when importing the results

maxwxyz, 2026-09-22:
fixed by https://github.com/FreeCAD/FreeCAD/pull/32866

closed 2026-09-22 by maxwxyz