FreeCAD/FreeCAD #18369: Assembly: part shifts when selecting a feature for reference in Insrt Part

prokoudine, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create an assembly and add a body to it.
2. Sketch a simple slot with two holes, then pad it.
3. Assembly toolbar > Insert a new Part.
4. Select one of the holes.
5. The entire body shifts, the LCS indicator stays where it is supposed to be.

https://github.com/user-attachments/assets/3c2865c7-e603-47b2-bdb7-87f36bdf15a3

### Full version info

```shell
OS: Fedora Linux 41 (Workstation Edition) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * sheetmetal 0.5.4
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-08:
@prokoudine does it also shift when the part is grounded? iirc you need at least one grounded part.
@PaddleStroke FYI

prokoudine, 2024-12-08:
Nice catch, yes, it doesn't move when the part is grounded. But then I start getting a "Link is broken warning" and a new part gets created in the root of the project tree rather than inside the assembly.

maxwxyz, 2024-12-08:
can you share the file?

PaddleStroke, 2024-12-16:
I see what is happening. The joint creation that happens in the insert part is trying to flip the first part When obviously it should not

closed 2024-12-20 by chennes