FreeCAD/FreeCAD #18371: holes make pads

Codich7, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

i create a pad for my object and the first one works and move to another sketch to make a hole and a pad appers. this works in reverse as well.such as putting a hole in first then the next sketch trying to pad i just get hol.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Curves 0.6.51
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-12-08:
Zip the FCStd file so you can push it here.

Codich7, 2024-12-08:
[NewZip.zip](https://github.com/user-attachments/files/18053992/NewZip.zip)

tarman3, 2024-12-09:
Does it bother you that the base model contains defects?

[image]
[image]

maxwxyz, 2024-12-09:
Seems like a broken file, does this also happen with a good model?

luzpaz, 2025-05-10:
@Codich7  can you reproduce from scratch ?

FEA-eng, 2025-09-17:
Indeed there are some artifacts in the model after recomputing:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-10-05:
[This](https://github.com/FreeCAD/FreeCAD/issues/18371#issuecomment-3302938928) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```