FreeCAD/FreeCAD #18372: FreeCAD v1.0 selects the wrong edges for chamfers and fillets when opening older FreeCAD files.

TadPath, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The problem is that FreeCAD v1.0/RC1, selects the wrong edges for filleting / chamfering in files made with older (pre v1.0) versions of FreeCAD when those files are opened with the latest RC1/v1.0 version.
This can manifest in two ways:
1. The models are shown without errors but are 'touched' (marked for re-calculation) and when recalculation is done FreeCAD applies the chamfer and/or fillet to the wrong edges (i.e. to edges that were not the ones originally selected for chamfering or filleting in the original FreeCAD file made with the older version of FreeCAD).
2. The models are shown with errors agains the fillet / chamfer operations and, opening those fillets / chamfers shows that the error is due to FreeCAD trying to apply those fillets / chamfers to wrong edges that were not the ones originally selected and which genuinely will be erroneous if the fillet / chamfer is applied to them.

In other words RC1/v1.0 FreeCAD wrongly assigns edges for chamfers and fillets for models created in older FreeCAD versions. Here are some examples:

[image]
[image]
[image]
[image]
[image]
[image]
[image]

You may download the FreeCAD files used for this from my Google Drive with [this link](https://drive.google.com/drive/folders/15nRBnC0SqlMsNcmElwT500wWt-M28eap?usp=sharing)

The pictures from top to bottom show:
1. What the problem looks like in RC1/v1.0 before fixing for fillets - the correct shapes are shown in the render but the model tree shows error
4. When opening the erroneous chamfer in RC1/v1.0 it shows not edges selected at all
5. Manually fixed by re-adding the edges
6. What the fillet problem looks like when opening in RC1/v1.0 - note the correct filleted shape is shown in the render but the model tree shows error on the fillet
7. The view when opening the erroneous fillet in RC1/v1.0 shows that edges are selected but these do not correspond to those edges shown filleted in the previous picture
8. Another view of the opened erroneous fillet
9. The view after deleted all selected edges and re-assigned them manually. The result is good (the render is as was shown in picture 4 and the error in the model tree gone after this - picture not shown).

The files in the Google drive folder are:
Pre_RC1.FCStd - the chamfer problem file before being fixed
Manually_fixed_RC1.FCStd - the chamfer problem file after manual fixing in RC1/v1.0
Pre_RC1b.FCStd - the fillet problem file before being fixed
Manually_fixed_RC1b.FCStd - the fillet problem file after manual fixing in RC1/v1.0

Note 1: ALL these files were saved with RC1 (I had to extract the models from very large files and save them in separate files for the purpose of making this issue).

Note 2: I cannot 'downgrade' back to pre RC1 to tell you exactly what version of FreeCAD they were made in but is was probably an 0.21 version).

The above are examples where FreeCAD shows an error in the model because the mis-selected edges can't take the chamfer / fillet so show up as an error but, perhaps more worrying / important is that in some models the wrongly selected edge CAN take the chamfer / fillet so NO ERROR is shown and when the model is re-calculated it develops the wrongs shape because of this.

Thanks.

### Full version info

```shell
OS: Arch Linux (LXDE/LXDE/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * fasteners 0.5.29
  * OpticsWorkbench 1.0.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-09:
So it does not happen with 0.21.2 files?
@CalligaroV FYI, not sure if relevant when created with dev versions?

CalligaroV, 2024-12-14:
@maxwxyz yes, in this case I would say that the cause of this issue isn't related to a specific dev revision but rather is a compatibility issue with `v0.21`.

I tried to open @TadPath's  [files](https://drive.google.com/drive/folders/15nRBnC0SqlMsNcmElwT500wWt-M28eap?usp=sharing) with a branch I'm working on for another issue (https://github.com/CalligaroV/FreeCAD/tree/toponaming-ElementMapVersion-code-from-LS3) but this isn't fixing this issue.

Tried also with LS3 but the outcome is the same, the issue is still present.

@TadPath, if I understood correctly the [files](https://drive.google.com/drive/folders/15nRBnC0SqlMsNcmElwT500wWt-M28eap?usp=sharing) you shared are made with a `v1.0 dev` dev version because you're no longer able to run v0.21 on your machine.
Am I right?

If so then it would be pretty hard for me to figure out why this issue is happening:
as wrote above, I think that this is a compatibility issue but to prove that (test with https://github.com/CalligaroV/FreeCAD/tree/toponaming-ElementMapVersion-code-from-LS3 and LS3) I need the original files made with `v0.21`.

If you can't share them can you try opening them with LS3 and give us feedback?
It may help with the analysis.

TIA!

TadPath, 2024-12-14:
Hi @CalligaroV , thanks for trying this. I am pleased to say I have made some progress with v.0.21.1
First, yes you are correct in saying the files I made available on my Google drive are saved in v1.0 because I can't get v0.21 to run on my machine.
However, I have a Windows 10 machine I rarely use nowadays but I booted it up today and managed to load the Windows version of FreeCAD v0.21.1 on it so I loaded my original big FreeCAD file (which I made in August 2024 with pre v1. FreeCAD) and extracted the same two models into separate files which I saved with that v0.21.1 and so now I attach zipped versions of those files here.
[v0211_FilletsOK.zip](https://github.com/user-attachments/files/18135939/v0211_FilletsOK.zip)
[v0211_ChamfersOK.zip](https://github.com/user-attachments/files/18135941/v0211_ChamfersOK.zip)
The one called ChamfersOK is the equivalent of the file Pre_RC1.FCStd on my Google drive.
The one called FilletsOK is the equivalent of the file Pre_RC1b.FCStd on my Google drive.

The only difference is that the Google drive files were models extracted from a pre v1.0 FreeCAD file into v1.0 FreeCAD and saved with v1.0 FreeCAD but the current zip versions were (the same) models extracted from the same pre v1.0 FreeCAD file using v0.21.1 Windows FreeCAD and saved with v0.21.1 Windows FreeCAD.

The version info from the Windows FreeCAD is:

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)

Interestingly, when I load the Pre_RC1.FCStd and Pre_RC1b.FCStd files into v0.21.1 FreeCAD it shows the errors described in this issue but the zip files provided here (that only every existed and were saved in a pre v1.0 FreeCAD) show no errors at all in v0.21.1 FreeCAD but do show the exact same errors described in this issue when loaded into v1.0 FreeCAD.

Here is a screenshot of the pre v1.0 files in v0.21.1 FreeCAD - no errors:

[image]

So this confirms to me that the issue has crept in at some point between v0.21.1 and v1.0.

I do not know how to install your branch version of FreeCAD to try it and I can't compile from source on my system as it stands so if you can link me to an AppImage I could try. Otherwise, now you have access to original unadulterated pre v1.0 FreeCAD files of my models to try for yourself.

All the best.

P.

CalligaroV, 2024-12-14:
Many thanks @TadPath for sharing the `v0.21` files!

I'm happy to inform you that, after testing yours [v0211_FilletsOK.zip](https://github.com/user-attachments/files/18135939/v0211_FilletsOK.zip) and [v0211_ChamfersOK.zip](https://github.com/user-attachments/files/18135941/v0211_ChamfersOK.zip) with my https://github.com/CalligaroV/FreeCAD/tree/toponaming-ElementMapVersion-code-from-LS3 the issue looks solved, at least in my local tests.
The only side effect looks to be an appearance issue that can be easily solved by selecting/resetting the material/appearance.

v0211_FilletsOK
[image]

v0211_ChamfersOK
[image]

As wrote in my https://github.com/FreeCAD/FreeCAD/issues/17124#issuecomment-2525716196, I'm going to open a PR with a modification that will fix that issue and, likely, also this one.
Therefore probably in the (near) future your issue will be solved.

About your request
> if you can link me to an AppImage

sorry but I haven't yet studied how to create AppImages and therefor for that I can't help you.

However we may ask to other users if someone can try your `v0211*` files with my branch.

@maxwxyz can you help us with this?
Otherwise I may dare to bother @FEA-Eng, asking him if he can perform the tests, as he already did that for #17124

Waiting for feedback.

Thanks to everyone!

TadPath, 2024-12-14:
Hi @CalligaroV , that looks fantastic. Thanks. Is there any way I can be notified when your fixes get merged into the main FreeCAD distro so I can try it out?
Out of interest, it says, when you open the models, that they need re-calculating. Have you tried to recalculate them? Does that create or show any errors after recalculating?

CalligaroV, 2024-12-15:
@TadPath

> Is there any way I can be notified when your fixes get merged into the main FreeCAD distro so I can try it out?

ATM I can only suggest you to check the commits applied to [main](https://github.com/FreeCAD/FreeCAD) from time to time, especially after monday's and friday's [PR Meetings](https://forum.freecad.org/viewtopic.php?t=80228).
IIRC After those meetings a new development build (and also AppImages) is generated, containing the last modifications applied.
You'll be able to download the AppImages from https://github.com/FreeCAD/FreeCAD-AppImage/releases/tag/weekly-builds

> Out of interest, it says, when you open the models, that they need re-calculating. Have you tried to recalculate them? Does that create or show any errors after recalculating?

Are you talking about the recordings in my previous post?
If so then absolutely yes!

When the recompute dialog appears I hit `Yes` (or better, `Y` on the keyboard :sweat_smile:) and the model is recalculated.
The output of the recalculation is what you can see on my previous post: models opened without errors (no error icons in the Tree View).

As you can see some warnings appear but those do no harm, at least not during the opening of the model.

Maybe those will produce some side effects while continuing the modeling, but if so then IMHO the right thing to do would be to open a new issue describing what's going on.

szdowk, 2025-01-23:
Hi

There is another problem with chamfer while opening 0.21.2 files in 1.0.0.

If option "Two distances" was used in 0.21.2 while creating of chamfer, parameters "Size" and "Size2" are swapped during interpretation when opening in 1.0.0. 
They are imported appropriately as "Size" and "Size2" but used vice versa.

Is this the same issue, or should be reported separately?

drwho495, 2025-10-05:
i've done some testing with a [version of 0.22 that did not have the toponaming code enabled](https://snapcraft.io/freecad-mechanical) and the errors were present. this likely isn't an issue with the toponaming solution, and it's something unrelated

FEA-eng, 2026-10-23:
[This](https://github.com/FreeCAD/FreeCAD/issues/18372#issuecomment-4241453624) is still relevant

[Pre_RC1.FCStd.zip](https://github.com/user-attachments/files/33855836/Pre_RC1.FCStd.zip)

[Pre_RC1b.FCStd.zip](https://github.com/user-attachments/files/33855837/Pre_RC1b.FCStd.zip)

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```