FreeCAD/FreeCAD #18375: Sketcher: Trim tool create malformed constraints in some cases

Shkolik, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trim tool working illogical... Simple operation to trim part of the circle results in malformed center coincident to origin constraint. See attached file and video.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CurvedShapes 1.0.8
  * Curves 0.6.28
  * fasteners 0.5.21
  * Foamcut 0.1.5
  * Marz 0.1.5
  * OpenTheme 2024.9.1
  * PieMenu 1.8.1
  * SmartReportFields1.xlsx
```

### Subproject(s) affected?

Sketcher

### Anything else?

Test file: 
[test_trim.zip](https://github.com/user-attachments/files/18054526/test_trim.zip)

<video src='https://github.com/user-attachments/assets/1b3e5ad7-2cdf-4783-b9e0-cde083855dda' width=600/>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-11:
Related: https://github.com/FreeCAD/FreeCAD/issues/17605, https://github.com/FreeCAD/FreeCAD/issues/12715

AjinkyaDahale, 2025-01-07:
#18665 does not have this problem, likely as a side effect of other changes. However, the symmetry of the rectangle is not preserved, and I predict it'll be difficult to make it so easily.

Could someone try to replicate this in 1.0.0 and report how it behaves?

FEA-eng, 2025-09-14:
Sometimes, I get this:

[image]

```
18:04:15  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Access violation
18:04:15  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').trim(5,App.Vector(2.643418,16.286876,0))
18:04:15  Unnamed#Sketch: Failed to trim edge
```

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

PaddleStroke, 2026-09-07:
I can't reproduce either

closed 2026-09-07 by maxwxyz