FreeCAD/FreeCAD #18376: [Preferences > Display >Advanced > Animation curve type] Unexpected text movement

Michkovy, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The text of the Animation curve type setting moves to the extreme right of the window when changing the setting. The text stays there until the settings window is reopened.

[image]

### Full version info

```shell
Issue persists in safe mode

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-12-10:
@kadet1090, you might want to take a look at this.

Frankleklek, 2025-03-07:
Hi! I'm interested in solving this issue. Can I work on it?

closed 2025-04-06 by kadet1090

luzpaz, 2025-04-06:
Thanks @Frankleklek!

PR #20553 (2025-04-06): Preferences: Fixed a bug in 'Animation curve type' label position
Deleted a line in DlgSettingsAdvanced.cpp that made the label move after selecting a animation curve type. Fixes #18376.
commit ceee2676 (2025-04-06): Preferences: Fixed a bug in 'Animation curve type' label position

Deleted a line in DlgSettingsAdvanced.cpp that made the label
move after selecting a animation curve type. Fixes #18376.