FreeCAD/FreeCAD #18382: Core: SoShapeScale does not take DPI scaling into account

PaddleStroke, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

SoShapeScale does not take DPI scaling into account.
So if the user has 200% scaling on his OS, the object appears twice smaller than it should : 

[image]

One of the problem of https://github.com/FreeCAD/FreeCAD/issues/18337

@wwmayer FYI

### Full version info

```shell
1.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-12-09:
One possible solution I see is to use SbViewportRegion::getPixelsPerInch to make a factor like this : 
```
void SoShapeScale::GLRender(SoGLRenderAction * action)
{
    SoState * state = action->getState();

    SoScale * scale = static_cast<SoScale*>(this->getAnyPart(SbName("scale"), true));
    if (!this->active.getValue()) {
        SbVec3f v(1.0f, 1.0f, 1.0f);
        if (scale->scaleFactor.getValue() != v)
            scale->scaleFactor = v;
    }
    else {
        const SbViewportRegion & vp = SoViewportRegionElement::get(state);
        const SbViewVolume & vv = SoViewVolumeElement::get(state);
        SbVec3f center(0.0f, 0.0f, 0.0f);
        float nsize = this->scaleFactor.getValue() / float(vp.getViewportSizePixels()[1]);
        SoModelMatrixElement::get(state).multVecMatrix(center, center); // world coords
        float sf = vv.getWorldToScreenScale(center, nsize);

        float pixelDensityFactor = vp.getPixelsPerInch() / 96;
        sf *= pixelDensityFactor;

        SbVec3f v(sf, sf, sf);
        if (scale->scaleFactor.getValue() != v)
            scale->scaleFactor = v;
    }

    inherited::GLRender(action);
}
```

But perhaps it would be best to use QT_SCALE_FACTOR so that the object scale synchroneously with the application. However I don't know how to read this variable.

PaddleStroke, 2024-12-09:
Interestingly the new LCS objects are scaling up when the height of the application is reduced. Which does not seem to happen with the SoAxisCrossKit even though it is also using SoShapeScale. So I'm a bit confused about this.
https://github.com/user-attachments/assets/73351257-5a9f-408b-83a4-925df1915a07

PaddleStroke, 2024-12-09:
Ok now I understand my confusion. The small coordinate system in the corner is not SoAxisCrossKit. It is something else.
You can draw the SoAxisCrossKit by doing : 
`Gui.ActiveDocument.ActiveView.setAxisCross(True)`
It looks like a document origin visualization.

PaddleStroke, 2024-12-09:
Ok so to fix this weird scale issue we just need to do 

`        float nsize = this->scaleFactor.getValue() / float(vp.getViewportSizePixels()[0]);`
instead of 

`        float nsize = this->scaleFactor.getValue() / float(vp.getViewportSizePixels()[1]);`

Then the object stays the same size whatever the viewport size.

Now back onto the topic. `vp.getPixelsPerInch` does not appears to work. As I get a 0.75 ratio on both screens. Perhaps this needs a software restart.

PaddleStroke, 2024-12-09:
Ok so I could fix it with : 

```
float getQtScaleFactor() {
    for(QWidget * widget : QApplication::topLevelWidgets()) {
        return widget->devicePixelRatio();
    }
    return 1.0; // Default scale factor if not set
}
SoShapeScale::GLRender(SoGLRenderAction * action)
{
    auto* scale = static_cast<SoScale*>(this->getAnyPart(SbName("scale"), true));
    if (!this->active.getValue()) {
        SbVec3f v(1.0f, 1.0f, 1.0f);
        if (scale->scaleFactor.getValue() != v){
            scale->scaleFactor = v;
        }
    }
    else {
        SoState* state = action->getState();
        const SbViewportRegion & vp = SoViewportRegionElement::get(state);
        const SbViewVolume & vv = SoViewVolumeElement::get(state);

        SbVec3f center(0.0f, 0.0f, 0.0f);
        float nsize = this->scaleFactor.getValue() / float(vp.getViewportSizePixels()[0]);
        SoModelMatrixElement::get(state).multVecMatrix(center, center); // world coords
        float sf = vv.getWorldToScreenScale(center, nsize);

        sf *= getQtScaleFactor();

        SbVec3f v(sf, sf, sf);
        if (scale->scaleFactor.getValue() != v){
            scale->scaleFactor = v;
        }
    }

    inherited::GLRender(action);
}
```

wwmayer, 2024-12-09:
Please do not add a dependency to Qt stuff here.

A better way is to implement a sub-class of SoFloatElement e.g. SoDevicePixelRatioElement where you set & get the device pixel ratio. As a reference implementation you can have a look at SoPointSizeElement that is very similar.

Then inside View3DInventorViewer when the rendering is done set the current DPR. Inside SoShapeScale::GLRender get the value.

Inside `SoDevicePixelRatioElement::initClass()` you may have to add this line `SO_ENABLE(SoGLRenderAction, SoDevicePixelRatioElement);` in case it fails to return the expected value.

PaddleStroke, 2024-12-09:
ok

closed 2024-12-10 by wwmayer

maxwxyz, 2024-12-18:
@PaddleStroke it works great for the datums itself, but the text label at the axis ends is not scaled now. When moving it between different dpr screens the text is smaller on the high dpi screen (half as big)