FreeCAD/FreeCAD #18383: PartDesign: Fillet: BRep_API: command not done - on a padded circle with a sphere hollow on its side

7th-fret, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Sketch a circle on XY and pad it to make a cylinder
2. Sketch on XZ a half circle closed by a line (the arc should be inside the mass of cylinder, set its position at the side of cylinder and groove cut (select vertical line of arc as groove axis) to make a sphere hollow.
3. Then select the side of cylinder and try fillet.
The fillet fails with Fillet: BRep_API: command not done
[image]

If on step 2 the arc is sketched outside of cylinder mass, the step 3 makes the fillet successfully.
[image]

### Full version info

```shell
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39512 (Git)
Build type: Release
Branch: main
Hash: d37115c692e66d3c5aa821eaec57e7b3dd0f1cd4
Python 3.12.7, Qt 5.15.16, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * A2plus 0.4.68
  * freecad.gears 1.3.0
  * OpticsWorkbench 1.0.26
  * OpenTheme 2024.5.16
  * sheetmetal 0.6.1
  * fasteners 0.5.33
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-09:
This kind of problem occurs frequently. Associated with the sphere linethe same issue 

[image]

7th-fret, 2024-12-12:
> Known OCC issue unfortunately. Could you add the example file with both sketches as .zip?

Yes, of course.

[examples.zip](https://github.com/user-attachments/files/18111199/examples.zip)