FreeCAD/FreeCAD #18385: Core: SoShapeScale + preselection broken

PaddleStroke, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

An object that has SoShapeScale can be preselected (status bar shows it is being preselected), however the graphic of the object does not show the preselection status.

To reproduce : 
Create body. 
Show origin object
Hover on top of it.

@wwmayer any idea how to fix this? Thanks

### Full version info

```shell
1.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-12-09:
Ok so it appears the SoFCSelection has to be outside of the SoShapeScale.

So it has to look like 

```
    auto highlight = new SoFCSelection();

    soScale->setPart("shape", sep);
    soScale->scaleFactor = 1.0;

    highlight->addChild(soScale);

    addDisplayMaskMode(highlight, "Base");
```

PaddleStroke, 2024-12-09:
Unfortunately with my currenty fix, we are loosing the hidden lines as dashed. All lines appear as solid.

Current fix : 
```
void ViewProviderDatum::attach(App::DocumentObject* pcObject)
{
    ViewProviderGeometryObject::attach(pcObject);

    // Create an external separator
    auto sep = new SoSeparator();

    // Add material from the base class
    sep->addChild(pcShapeMaterial);

    // Bind same material to all part
    auto matBinding = new SoMaterialBinding;
    matBinding->value = SoMaterialBinding::OVERALL;
    sep->addChild(matBinding);

    // Setup font size
    auto font = new SoFont();
    static const float size = App::GetApplication().GetParameterGroupByPath("User parameter:BaseApp/Preferences/View")->GetFloat("DatumFontSize", 15.0);
    font->size.setValue(size);
    sep->addChild(font);

    // Create the selection node
    auto highlight = new SoFCSelection();
    highlight->applySettings();
    if (!Selectable.getValue()) {
        highlight->selectionMode = Gui::SoFCSelection::SEL_OFF;
    }
    highlight->objectName = getObject()->getNameInDocument();
    highlight->documentName = getObject()->getDocument()->getName();
    highlight->style = SoFCSelection::EMISSIVE_DIFFUSE;

    // Visible features
    auto visible = new SoAnnotation();
    // Style for normal (visible) lines
    auto style = new SoDrawStyle();
    style->lineWidth = lineThickness;
    visible->addChild(style);

    // Visible lines
    visible->addChild(pRoot);

    // Hidden features
    auto hidden = new SoAnnotation();

    // Style for hidden lines
    style = new SoDrawStyle();
    style->lineWidth = lineThickness;
    style->linePattern.setValue(0xFF00); // (dash-skip)
    hidden->addChild(style);

    // Hidden lines
    hidden->addChild(pRoot);

    visible->addChild(hidden);

    sep->addChild(visible);

    // Scale feature to the given size
    float sz = App::GetApplication()
        .GetParameterGroupByPath("User parameter:BaseApp/Preferences/View")
        ->GetFloat("LocalCoordinateSystemSize", 1.0);  // NOLINT

    soScale->setPart("shape", sep);
    soScale->scaleFactor = sz;

    highlight->addChild(soScale);

    addDisplayMaskMode(highlight, "Base");
}
```

I guess that hidden must be added to the SoFCSelection for it to work.

So now I'm not sure how to fix this

wwmayer, 2024-12-09:
> Unfortunately with my currenty fix, we are loosing the hidden lines as dashed. All lines appear as solid.

Which hidden lines?

wwmayer, 2024-12-09:
OK, I see what you mean.

A bit off-topic but there are two issues:
* The origin point should be independent of the zoom level like the lines. If you try to create a very small object (e.g. a box with 0.5mm x 0.5mm x 0.5mm) it's impossible to see it if the coordinate system is shown.
* Trying to manually hide the origin and the planes doesn't work. In the tree view they are marked is invisible but in the 3D view they are still displayed. The visibility of the lines works as expected.

PaddleStroke, 2024-12-09:
I cannot reproduce the 2 issues you are mentioning.

https://github.com/user-attachments/assets/19bbe32b-d5df-452f-b14d-c900b1da1a8a

wwmayer, 2024-12-09:
Create a Part container or a body object. Show the origin. Start to zoom in and you will see the size of the point changes and that of the planes, too.
The axes stay the same.

PaddleStroke, 2024-12-10:
I cannot reproduce this. I'm currently building on top of 18332+18393 so it may be why. Perhaps it is platform specific? I have inquired other users to see if someone else can reproduce.

Aside from this do you have any clue about the SoShapeScale + preselection issue?

wwmayer, 2024-12-10:
The PR https://github.com/FreeCAD/FreeCAD/pull/18126 contained some code defects because it has overloaded the virtual method getRoot() of the view provider base classes (`getRoot()` vs `virtual getRoot() const`).
For me this caused the compiler warning `Gui::ViewProviderDatum::getRoot' hides overloaded virtual function [-Werror,-Woverloaded-virtual]` and because of the -Werror flag the build stopped with this error.

The solution to fix the problem is to rename `ViewProviderDatum::getRoot` to e.g. `ViewProviderDatum::getDatumRoot` but here I missed all occurrences of getRoot and that's why it caused the observed problems.

After finding the missing two occurrences in ViewProviderPlane and ViewProviderPoint both problems are fixed now.
See https://github.com/FreeCAD/FreeCAD/pull/18340

closed 2024-12-16 by PaddleStroke