FreeCAD/FreeCAD #18388: PartDesign: Cant create revolution

rmx-it, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
cant create a revolution for 60° around Sketch.Edge1.
The edges of the sketch are not visible as a reference.

[ClipStickv8.zip](https://github.com/user-attachments/files/18062644/ClipStickv8.zip)

[image]

Thats a regression as it worked in the past.

[image]

### Full version info

```shell
OS: FreeBSD 14.1-RELEASE-p6 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.1.0.0
Build type: Release
Python 3.11.10, Qt 5.15.16, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * lattice2 1.0.0
  * Assembly4 0.50.15
  * CurvedShapes 1.0.10
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-09:
There are errors in Sketch

[image]

rmx-it, 2024-12-09:
Hi,
sorry I dont see it.
Checking the sketch shows no error too.

I created the sketch as in the past:
- Create it on top of the upper face.
- Put in the edges of the upper face as external geometry.
- Create rectangular fully constrained using the edge points.

Then in the past I used this sketch to revolve around the one edge without a datum line.

tarman3, 2024-12-09:
https://github.com/user-attachments/assets/8d3fc6aa-715c-40fc-a413-e27e33a5875c

tarman3, 2024-12-09:
https://github.com/user-attachments/assets/70d8ff49-88dd-4a04-b667-d09e2de38c09

rmx-it, 2024-12-09:
Hi,
AH!
Now I know, how it happend:

If pulling in an line as extarnal geometry it got not pulled in as construction geometry.
So I had additional lines which led to a non closed geometry!

Did this behavior change in FreeCAD 1.0?
And is this a bug or feature?

rmx-it, 2024-12-09:
Hi,
and is it a bug or feature that I can't choose an edge of the sketch as a reference for the revolution
only a datum line?

tarman3, 2024-12-09:
Also you can use sketch `halterProfilOben` two times, for Additive Pipe and Revolution