FreeCAD/FreeCAD #18389: Core: Referring to parent properties not user-friendely

pieterhijma, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD disallows creating cyclic dependencies. This means that it is not possible to refer to properties from parent objects. 

FreeCAD has the hidden ref construct that allows users to hide an expression that refers to a parent object from the dependency check mechanism. However, in my humble opinion, this is not user-friendly and can lead to problems. See a discussion on the matter in [this forum topic](https://forum.freecad.org/viewtopic.php?p=788072#p788072) and beyond that.

### Full version info

```shell
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.39530 (Git)
Build type: Release
Branch: main
Hash: b1119ccc4ff30297de13098676c6f430e33a8b7d
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2024.5.16
  * osh-autodoc-workbench 0.2.3
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.29
  * WebTools 1.0.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pieterhijma, 2024-12-10:
@maxwxyz, I think the lack of user friendliness comes from deeper issues than just user interface, so I think the CWG is more applicable for this type of issue than the DWG (although the DWG is invited for their opinion as well, of course).

maxwxyz, 2024-12-10:
@FreeCAD/cad-advisory-group ^

kaiwas, 2024-12-11:
I completely agree that this point should be reworked.
1. A cyclic dependency is not always a truly cyclic dependency.
The fact that data moves in two directions (from parent to child or from child to parent) does not mean that the dependency is always cyclic.
2. In general, one could agree with this behavior (I'm talking about the "href" wrapper). Entering such a wrapper requires the user to spend effort and in itself can be a kind of protection. When the user wraps data, he knows exactly what he is doing. BUT! After entering the wrapper, it would be better if the tooltips worked the same as without the wrapper.

furgo16, 2025-04-20:
For future reference for folks landing on this issue, here's the `href` documentation => https://wiki.freecad.org/Expressions#Cyclic_dependencies