FreeCAD/FreeCAD #18391: BIM: 1.1dev RMB context menu IFC-> Load shape does not work

semhustej, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- Open sample file (remove the .zip file format)
[wall-file.ifc.zip](https://github.com/user-attachments/files/18064855/wall-file.ifc.zip)

- open the IFC file with these settings:
[image]

- right click the wall in the project tree and select IFC-> Load shape
- the shape is not loaded

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

IFC-> Load representation works as expected

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-12-09:
Tested also with 1.0 version and the bug is present there too.

maxwxyz, 2024-12-09:
@yorikvanhavre FYI

PR #18471 (2024-12-13): BIM: Fixed shape loading
NativeIFC objects shape would not load when ShapeType is None

Fixes #18391
closed 2024-12-13 by chennes