FreeCAD/FreeCAD #18394: Compatibility for renamed 'movePoint' to 'moveGeometries'

maxwxyz, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PR https://github.com/FreeCAD/FreeCAD/pull/18273 reamed 'movePoint' to 'moveGeometries' so this must be considered for compatibility.
@PaddleStroke FYI

### Full version info

```shell
https://github.com/FreeCAD/FreeCAD/pull/18273
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-12-10:
Well that wasn't long. CI is failing on my core datum label because of that : 
```
  File "/usr/local/Mod/PartDesign/PartDesignTests/TestTopologicalNamingProblem.py", line 2381, in testPD_TNPSketchPadSketchSplit
    doc.Sketch.movePoint(4, 1, App.Vector(10,2,0), False)
AttributeError: 'Sketcher.SketchObject' object has no attribute 'movePoint'
```

PaddleStroke, 2024-12-10:
Ah yes apparantly the test fails if you do not rebase your PR. Interesting.

PaddleStroke, 2025-03-10:
Shouldn't this be closed @maxwxyz ? I don't see something to be done.

PaddleStroke, 2025-03-10:
I cannot unassign myself from an issue? Please remove me thanks.

closed 2025-03-10 by maxwxyz