FreeCAD/FreeCAD #18395: Core datum: Plane can lose transparency in some conditions

PaddleStroke, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

If planes are the only visible object they becomes full as in above picture.
Reproduction steps : 
New doc.
New body
Show the origin object.
Hide the point and the 3 axis
-> planes become as in the picture.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-12-10:
This is happening because of the application of transparency to the lines of the planes : 

https://github.com/FreeCAD/FreeCAD/blob/16ac5f99770e078f97e07d7bcf06654f90846673/src/Gui/ViewProviderPlane.cpp#L76

PaddleStroke, 2024-12-10:
@wwmayer there also this bug which is leading me in circles. I tried to remove pretty much everything from ViewProviderDatum and ViewProviderPlane but could not find the root cause.
Perhaps it will ring a bell to you?

wwmayer, 2024-12-10:
No idea what's wrong.

PaddleStroke, 2024-12-11:
On further test, I see the same bug happens with ImagePlanes : 

https://github.com/user-attachments/assets/b3a934ac-9820-4ec5-99c9-3b96adce5d42

luzpaz, 2025-04-06:
Anyone able to further diagnose the bug ?

FEA-eng, 2025-10-28:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

luzpaz, 2026-10-10:
@Mikarisar can you please add your full [About](https//wiki.freecad.org/About) info that you tested against ?

luzpaz, 2026-10-10:
> I installed the OpenTheme addon and enabled OpenTheme in settings - planes and constraints icons now look normal, with transparency.

@obelisk79 would this indicate that the defualt FreeCAD themes are broken ?

obelisk79, 2026-10-10:
I'm not 100% sure that its a theme issue. 
I'm unable to reproduce the issue I just checked by starting in safe mode and followed the steps to reproduce outlined above.
Perhaps its a gpu driver issue? Or something OS specific?

[image]