FreeCAD/FreeCAD #18397: Feature Request: Enable Image Measurement Support via Image Plane Adjustment

MauricioLucas, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Title**: Feature Request: Enable Image Measurement Support via Image Plane Adjustment

**Description**:  
Currently, the measurement tool in FreeCAD works well across all workbenches, but it is not available for geometries in the **Image** module. When working with images, such as technical drawings or scanned floor plans, users need an easy way to measure distances or adjust the image scale accurately within FreeCAD.

### Problem Description:
When attempting to use the measurement tool with an image, the following error message appears:

```
No measure handler available for geometry of module: Image
```

This limits the workflow significantly, as users must resort to alternative methods (such as creating a reference geometry in the **Sketcher** workbench) to estimate distances on an image. This process is time-consuming and complicates the workflow, especially for users less experienced with FreeCAD.

### Suggested Solution:
Rather than relying on the "Measure" tool, a more efficient solution would be to introduce a proportional adjustment option in the "Image Plane Settings" menu. The user could specify a known distance between two points on the image, and then click to select and move two points on the image to adjust its scale. The image would then auto-adjust to the correct proportions, making it easier to work with the image at its proper scale.

This would simplify the workflow and reduce the number of steps required to use images accurately in FreeCAD. Users would not have to create additional sketches to scale their images properly, which is especially beneficial for beginners and non-technical users. 

### Benefits:
- **Improved User Experience**: A more intuitive way to work with images directly.
- **Efficiency**: Less time spent on setting up workarounds with additional reference geometry.
- **Access for Beginners**: Helps new users avoid complex steps when working with images in FreeCAD.

This solution could greatly enhance the workflow for a wide range of users, making FreeCAD more accessible and efficient for various applications, such as technical drawings, floor plans, and design reference images.

### Full version info

```shell
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Assembly4 0.50.15
  * 3D_Printing_Tools
  * LCInterlocking
  * PieMenu 1.10.1
  * Gridfinity 0.6.0
  * sheetmetal 0.5.10
  * WB_Organizer 2024.1.29
  * DynamicData 2.70.0
  * fasteners 0.5.32
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Glass

---

OS: Linux Mint 21.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Assembly4 0.50.15
  * 3D_Printing_Tools
  * LCInterlocking
  * PieMenu 1.10.1
  * Gridfinity 0.6.0
  * sheetmetal 0.5.10
  * WB_Organizer 2024.1.29
  * DynamicData 2.70.0
  * fasteners 0.5.32
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Glass
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Shkolik, 2024-12-10:
I think Image already have calibration tool that does exactly what described. 
Import image, double click it to enter edit mode.

[image]

Click on Calibrate button and then select 2 known points and set known distance:

[image]

Click Ok on task panel. DONE.

But it would be great to have not only scale calibration, but also rotation. Like select 2 points and set it horizontal or vertical, or any angle of your choice.

MauricioLucas, 2024-12-10:
_Hi, Andrew_

In a way, I feel embarrassed for not noticing this.  
I even saw the **"Calibrate"** button, clicked it, and it did nothing.  
Part of the blame for this lapse is the fact that hovering over buttons and options rarely shows the help popup, and when it does, it lasts less than 2 seconds. This is probably the bug I should report.  
_(These help descriptions could also be displayed in the status bar, similar to how it works in Blender.)_  

On the other hand, the term **"Calibrate"** isn't very clear in describing this function. I believe **"Proportion adjustment"** would make it more obvious.  

Thank you for enlightening my ignorance.

Shkolik, 2024-12-10:
[Mauricio](https://github.com/MauricioLucas), that's fine - you was 99.99% close to the answer :) I agree - task panel for the image could be more user friendly and self-explanatory than now.