FreeCAD/FreeCAD #18402: Cannot add length dimension from Spreadsheet through Expression editor

aaannorst, 2024-12-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi, I am trying to add dimensions to some objects on my sketch in a Part Design sketch through Expression editor from a Spreadsheet, but after selecting the variable, the old, hand-drawn dimensions remain on the sketch. 

ubuntu 22.04.05, Freecad 1.0.0
[image]
[image]
[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1202
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-12-10:
I think the example file would need to be provided to diagnose this, but it looks like a value is coming back from the spreadsheet, just not one that can be used as a measurement. 

Are you sure the correct spreadsheet cell has the alias AsztallapSzelesseg?

tarman3, 2024-12-10:
Can not reproduce

[image]

<details>
  <summary>FreeCAD 1.1.0dev.39560 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

luzpaz, 2025-01-25:
@aaannorst can you try to use the snap edge build and try to reproduce ?

See https://wiki.freecad.org/Ubuntu_Snap#Changing_between_different_Snaps

FEA-eng, 2025-09-14:
@aaannorst Can you provide an example file?

luzpaz, 2025-11-15:
No response from OP who was using 1.0.0

closed 2025-11-15 by luzpaz